Makuna/NeoPixelBus #151: NeoEsp8266UartMethod.cpp:34:22: fatal error: uart.h: No such file or directory

kb1sph, 2017-01-08:
For some reason this keeps happening when I use the suggested method of installing the esp8266 board by putting this link in the addition boards manager URLs.

http://arduino.esp8266.com/stable/package_esp8266com_index.json

But, if I take that link out and just download everything and put it under my hardware directory, it seems to compile fine.  I'm not sure if this is something that can be accounted for somehow in the library, or if it can be noted maybe in the instructions to download and install the board files manually.

Unfortunately, when I install the files manually into the hardware directory, it seems to be missing the tools\esptool.exe.

NOTE: I'm using Windows with Arduino v1.6.12

Makuna, 2017-01-08:
The file "uart.h" that it is looking for should be located at "C:\Users\<your user id>\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\"

If it is not then your system is not correctly configured.  This can happen due to previous bugs in Arduino IDE.  There is nothing I can do make this more robust.

With the Arduino IDE, you really need the latest version anyhow, which is 1.6.13.  They have had some strange bugs around library and board updates for the last year and have put out updates almost every month).

If you have been updating the Esp over the last year I suggest to uninstall Arduino IDE, manually delete the Arduino folders (the root of the one I list above especially, starting at Arduino15) and then install all the new stuff.

closed 2018-08-16 by Makuna

raymiec, 2018-11-19:
This one is easy, i was kicking myself when i discovered the problem.

Select your port in the tools menu, simple as that, strange as it seems, but it genuinely gave me that error til i did.

hussainb, 2018-12-15:
> This one is easy, i was kicking myself when i discovered the problem.
> 
> Select your port in the tools menu, simple as that, strange as it seems, but it genuinely gave me that error til i did.

Isn't working for me