Makuna/NeoPixelBus #152: ESP32 support...

sticilface, 2017-01-10:
You probably knew this would come.... :) ESP32 support

any chance?

Makuna, 2017-01-10:
I haven't received my Esp32 yet; but it is on the list.

sticilface, 2017-01-10:
:) I just ordered mine from someone in london. not cheap, but same as getting it from china!

sticilface, 2017-01-10:
so much RAM!

AdySan, 2017-02-11:
It's been a month, so just checking, I'd really appreciate if this library would work with esp32 arduino port, or better esp-idf itself!

Makuna, 2017-04-05:
Actively working on this now.

AdySan, 2017-04-05:
Yay!

Makuna, 2017-04-06:
https://github.com/Makuna/NeoPixelBus/pull/167
This adds bitbang method as the default, which can support pins below 32.
Hardware support is next

Snigelson, 2017-05-03:
Do you know yet which pin it will be on?

Makuna, 2017-05-03:
Hardware support will be able to also work on pins below 32.  There are at least two outstanding issues on the esp32 core iot libraries that need to be addressed before I can publish the hardware support; otherwise the "shared" RMT hardware can only be used by one system at a time (NeoPixel, IR receive, IR transmit, only one at a time, etc).  

Current work is waiting on the following two Esp32 IOF issues.
https://github.com/espressif/esp-idf/issues/514
https://github.com/espressif/esp-idf/issues/517

Buffalchill, 2017-05-11:
good to hear you are working on it ! Realy would like to see the ESP32 do the Neopixel while receiving data over wifi

Makuna, 2017-05-11:
@Buffalchill Have you tried the current Esp32 support while doing WiFi?  Theoretically it should work as the Esp32 is dual core.  Its just going to take a lot of the CPU resources.

Buffalchill, 2017-05-12:
@Makuna since yesterday i am trying out the ESP8266 with your WS2812B driver. It seems to work better than the Neopixel.h  Thumbs up!

When i compile it for the ESP32 i get this ERRROR:

> "In file included from C:\Users\Stefan Schmidt\Documents\Arduino\hardware\espressif\esp32\libraries\LiOr WiFi Projekt\Empfaenger\ESP32_WS2812B_WiFi_17_05_11\ESP32_WS2812B_WiFi_17_05_11.ino:65:0:
> 
> C:\Users\Stefan Schmidt\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:74:2: error: #error "Platform Currently Not Supported, please add an Issue at Github/Makuna/NeoPixelBus"
> 
>  #error "Platform Currently Not Supported, please add an Issue at Github/Makuna/NeoPixelBus"

Makuna, 2017-05-12:
@Buffalchill Standard data needed: What device/board are you selecting?  What version of NeoPixelBus library is installed?  Are you using the Arduino IDE or something else?  What version is the IDE?

Jorgen-VikingGod, 2017-05-12:
I just tried the lib (=master) with my ESP32 devboard and is running fine

Buffalchill, 2017-05-12:
Hi!  I had an older Lib. now it runs.
But with the ESP8266 my rainbow fader runs well, with the ESP32 it is fading but also blinking pretty strange :-(

Jorgen-VikingGod, 2017-05-12:
@Buffalchill did you connect it directly on GPIO or through level shifter?
To get stable signals it is always better to serve 5V data line. And of course a 100-220 Ohm resistor is also a good idea as "low-pass" filter to cut the noise.

PedroScreamerovsky, 2017-05-12:
I'm also having some problems with the ESP32. I'm using a NeoPixel Jewel, and the 4th led is blinking green, with all the example sketches. I've tried with a level shifter but still no luck. Also adding the delay didn't resolved the problem

Makuna, 2017-05-12:
@PedroScreamerovsky What actual LED model are you using?  WS2813 will require a specific method.

Buffalchill, 2017-05-12:
I got the same issues with the ESP32 like PedroScreamerovsky.
Now i added a level shifter and a resistor but this didnt realy help. With the ESP8266 the code runs pretty well, but on the ESP32 its flickering on the whole stripe :-/

PedroScreamerovsky, 2017-05-12:
I'm using these: https://www.adafruit.com/product/2859 (grbw)
I've tried Neo800KbpsMethod, then NeoEsp32BitBang800KbpsMethod (with all the pixels working well except one, that has the green channel doing weird things), and finally the NeoEsp32BitBang400KbpsMethod, that resolves the issue of the green led, but creates lot of random flickering.
As for Buffalchill, the esp8266 is working fine with the same sketch

Makuna, 2017-05-12:
One more piece of information, how many are in strip?
(the hardware version is pending some improvements from espresif)

PedroScreamerovsky, 2017-05-13:
It's 7 leds

Buffalchill, 2017-05-13:
150 leds... pretty much flickering on the stripe

sebastius, 2017-05-17:
NeoEsp32BitBang800KbpsMethod

NeoEsp32BitBang800KbpsMethod supports any available pin below 32.

Hi, why 'any pin BELOW 32'? Guess who has a board with SK6812 RGBWs on pin 32...

Makuna, 2017-05-20:
@sebastius Pin 32 and above are handled differently.  The hardware version will use the built in pin matrix switching (allowing hardware to use more pins unlike the esp8266) which also doesn't support 32 and above.

Makuna, 2017-06-07:
BTW, I just read that the newer WS2812b coming off production are the same as the ws2813, so if the delay after calling show fixes it for you, switch to use the ws2813 method and the delay should not be needed anymore.  They changed the spec so it requires a longer reset time, but didn't change the model number.

r-neves, 2017-06-11:
I got the same issues with the ESP32 like Buffalchill.
Now i added a level shifter and a resistor but this didnt realy help. With the ESP8266 the code runs pretty well, but on the ESP32 its flickering on the whole stripe :-/

I am using this for the ESP32: 
NeoPixelBus<NeoGrbFeature, NeoEsp32BitBang800KbpsMethod> strip(PixelCount,13);
i am using the WS2812b stripes

ESP32 as some port like ESP8266 that works fine, because using ESP8266 has especific port otherwise strip flickering too.

Makuna, 2017-06-11:
use this instead...
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBangWs2813Method> strip(PixelCount,13);
```

PRO2XY, 2017-06-19:
Hi,
I was testing an RMT based WS2812 library for ESP32, which works pretty fine for me (without any level shifter). Maybe it could be of any help to this project?

https://github.com/FozzTexx/ws2812-demo

Makuna, 2017-06-19:
@PRO2XY If you read the above, this is what is planed.  I have two bugs I need fixed before I can move forward.  Their are two reasons why the RMT that is implemented today are not functional longer term without these.  They are being fixed/applied to the code; but its not ready yet.

1) The RMT format is extremely verbose for NeoPixels.  For strips of any size the memory use is terrible; requiring 4 bytes for every bit of data; or an average pixel would require 96 bytes to represent.  To solve this requires translation on the fly in the RMT ISR.
2) The RMT ISR is not exposed in such a way to be able to share it.  So someone using RMT could only use it for one output, so no IR and NeoPixels together.  That's a waist of the other seven channels of the RMT.

r-neves, 2017-06-24:
Now I am using a SK6812. With the ESP32 an its flickering on the whole stripe :-/

I am using this for the ESP32:
NeoPixelBus<NeoGrbwFeature, NeoEsp32BitBang800KbpsMethod> strip(PixelCount,13);
i am using the WS2812b stripes

ESP32 as some port like ESP8266 that works fine, because using ESP8266 has especific port otherwise strip flickering too.

How can fix it and what is the correct pin to use?

Makuna, 2017-06-25:
@r-neves I believe I gave you something to try above, but you may have missed it.
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBangWs2813Method> strip(PixelCount,13);
```
Some of the new chips are using the newer Ws2813 protocol.

r-neves, 2017-06-27:
But my strip is SK6812, and stills flicking.
If I turn Red and Blue flicker, but Green and White no flicker

Makuna, 2017-06-28:
@r-neves Are you stating that the simple test (without WiFi) the problem also exists?  If so, please join on the gitter channel to diagnose your issue, as its not the same as what this issue is tracking.

r-neves, 2017-06-28:
My code is very simple
`#include <NeoPixelBus.h>

const uint16_t PixelCount = 144; // make sure to set this to the number of pixels in your strip
NeoPixelBus<NeoGrbwFeature, NeoEsp32BitBangWs2813Method> strip(PixelCount, 14);

void setup()
{
  strip.Begin();
}

void loop()
{
  for (uint16_t i = 0; i < strip.PixelCount(); ++i)
  {
    RgbwColor color = RgbwColor(0, 255, 0, 0);
    strip.SetPixelColor(i, color);
    strip.Show();
  }
}`

Makuna, 2017-06-28:
@r-neves This "issue" had changed to tracking hardware support, I am sorry this was not obvious and I changed the title to make it so.

Please do as I ask and join the gitter channel for support **for your specific problem**; as often there are problems with connectivity and powering the NeoPixels that cause most of these flickering issues.

phenzao, 2017-08-15:
I am using the same example like r-neves above with esp32 dev board and platformio and I get this error: "undefined reference to `bitbang_send_pixels_800'"

EBZ-Krisemendt, 2017-09-06:
stumbled across this issue while working on a similar problem with faulty pixels on esp32.
as a workaround (while hardware-support isnt ready) adding the delay(1) right before strip.Show() as descripted in issue https://github.com/adafruit/Adafruit_NeoPixel/issues/139 help me to resolve the problem and all pixels are showing correctly.

Makuna, 2017-09-06:
Thanks for the reference.  I was afraid of something like this coming along, and with this knowledge I will make the hardware support the default in the future. 

I really wish they would fix the RMT support so I can finish the hardware version and get it out for people to use.

EBZ-Krisemendt, 2017-09-18:
is it to soon to ask for any updates already since the related issue https://github.com/espressif/esp-idf/issues/514 got closed? ;)

Makuna, 2017-09-18:
I still need [espressif/esp-idf#517](https://github.com/espressif/esp-idf/issues/517) fixed.

Makuna, 2017-10-08:
Related issue https://github.com/Makuna/NeoPixelBus/issues/182 to this conversation; I will be updating the bitbang with a fix soon while I am still waiting for a fix from the SDK as linked above for the hardware support.

chrisella, 2017-12-06:
Hey,
Any news on this?
I have a 240 pixel strip which works almost fine using a combination of the delay(1) and NeoPixelBus<NeoGrbFeature, NeoEsp32BitBangWs2813Method> strip(PixelCount, PixelPin);

The only problem is 1-2 pixels don't behave and are permanently turned on, usually earlier in the chain in-case that has any bearing. This is using the ESP32 without any kind of WIFI features, as soon as I enabled that I began having all manner of problems again with the whole strip flickering etc.

Makuna, 2017-12-06:
I still need espressif/esp-idf#517 fixed.
Please go to the above issue and up vote it.  It has been pretty stagnate for awhile.

chrisella, 2017-12-06:
Thanks, will do, thanks for continuing to push the ball on this!

vifino, 2018-01-09:
With the git version and `NeoPixelBus<NeoGrbFeature, NeoEsp32BitBang800KbpsMethod> strip(LEDS, PIN);` my 256 led ws2812b matrix has minimal errors (ie pixel set to weird/random value) and no flickering.
Happens less often than once a second when updating everything at ~60hz.

Before, I didn't have the git version and didn't select the bit banging method. It was unusable: Very heavy flicker and lots of pixel corruption.

mcer12, 2018-02-15:
Seems like things started moving on espressif side. @Makuna

Makuna, 2018-02-15:
Yes!

alexkirillov, 2018-03-06:
ESP32-digital-RGB-LED-drivers, mentioned earlier by PRO2XY is the only (arduino, didn't try any esp-idf) library that I was able to make work with my ESP32 and 3x 241 pixel LED rings. 

All other libraries, unfortunatly including NeoPixelBus at the moment, give arbitrary timing errors.
Althru they are rare, like 1 wrong pixel in 1000.
But given the setup is 3x 241, it occures every second time I'm setting lights to rgbw(0,0,0,0)

i understand that @Makuna is waiting for @igrr to optimize RMT stuff, not to waste all the memory on pixel data.

JanVelleman, 2018-03-19:
Hi Makuna,
Any news about this?

Sincerely

iontichy123, 2018-03-23:
Also interested in this topic. Any plans?

Makuna, 2018-03-23:
I still need espressif/esp-idf#517 fixed.

r1dd1ck, 2018-05-24:
Just to let everyone know - the required changes have been recently merged with the IDF master, so it is now only on the folks maintaining the **`arduino-esp32`** package to update it. 

Could be days, but could also be months - hard to tell :-s But I think they are currently already behind schedule with IDF updates, so maybe it will be sooner than later :-)

PS: I have the changes already incorporated into NeoPixelBus and currently running 8-way parallel without breaking a sweat, so good news indeed :-)

Makuna, 2018-05-24:
I also have a i2s implementation ready (based on the esp8266 DMA) but the esp32 team has asked me to hold off publishing it so that they can wrap the guts of it into an API.

MichMich, 2018-06-07:
Hi Guys! Any status update on this? I'm really in the need of a fix. Thanks!

r1dd1ck, 2018-06-09:
@MichMich 
The **RMT** implementation will need to wait till [esp-idf@5bc8ae3](https://github.com/espressif/esp-idf/commit/5bc8ae3b3a8be3e276c3318865efa9f0b5d82720) hits the `arduino-esp32` core. You can go to the [arduino-esp32](https://github.com/espressif/arduino-esp32/) repo and bugger them about it - which might speed things up a bit (or not) :-D

The **I2S** implementation mentioned by @Makuna, on other hand, would probably not require any changes to the current core, so it should be ready to go. But only in case the author is nice enough to share a pre-release version, of course ;-)

MichMich, 2018-06-12:
@Makuna, is it possible for you to share the i2s solution?

Makuna, 2018-06-12:
@MichMich The good news is that I have been given the go ahead on the i2s implementation; so it should show up within a few days.

MichMich, 2018-06-12:
@Makuna Awesome! Looking forward to it! :)

Makuna, 2018-06-13:
NEW UPDATE COMING:  I just merged in Esp32 Hardware Support using I2S (very similar to how Esp8266 was done but without the pin restrictions).  You can access it today by using Github to get the latest code.  A new library (through the Arduino IDE) will be made later today and might take a day or two to show up.

The new default on Esp32 will be this I2s method, using the i2s bus 1.

This ISSUE will remain open tracking RMT hardware support.

[27 more replies]

closed 2019-06-12 by Makuna