Makuna/NeoPixelBus #153: Changing brightness is changing the RGB color

bcatalin, 2017-01-13:
Still don't know if is really a bug or I didn't understood the library.

Part of a bigger project I need to modify the color of a strip  and its brightness.

Setting the stip to a color ( R or G or B) and changing the brightness is  fine but if I set the strip to
RGB (128, 128, 128) and change the brightness  for example increment/decrement it in 1 unit steps like 
```
setb = strip.GetBrightness() + 1 * delta;   
strip.SetBrightness (setb);  
strip.Show(); 
```

I am ending up in a blank strip 

Set bri: 59
Set bri: 58
Set bri: 57
Set bri: 56
Set bri: 55
R=0 G=0 B=0

Makuna, 2017-01-14:
Please include a full sketch that demonstrates the problem; along with the debug output.  What you gave me is not enough to understand or see the problem.  

NOTE:  Once the internal color channel reaches zero, it will always be zero.  So you can never expect to set brightness to zero and then brightness to 100 and expect anything other than black.  Dimming this way is a destructive operation (original color is lost).

The only way to retain the original color and do dimming is to use the linear blend method on the color; blend between the color and black by the amount you want and then set all the pixels to that color.

Makuna, 2017-01-14:
I am not sure the snippet of code you provided is doing what you want.
```
setb = strip.GetBrightness() + 1 * delta;  
```
The order of mathematical operation makes this (delta * 1) and then plus the current strip brightness.

Makuna, 2017-01-14:
I did some analysis of the code (the brightness feature is Adafruit's code ported, to remain compatible with them).
An iterative approach like you are doing for dimming is causing the small error due to math on every change to accumulate.  If you iteratively dim by 2, you get different values.  Again iterate by 3 even further different results.   This is the nature of the math for their implementation.

This is why I originally didn't include it at all, its flawed.

The best approach is like I stated, keep the original values, calling the RgbColor.LinearBlend() to blend to black, passing a "dim" value of 1.0 (full brightness) to 0.0 (full dim) and then pass the results to the SetPixelColor.

closed 2017-01-14 by Makuna