Makuna/NeoPixelBus #155: ESP8266 UART: RGB 255 isn't constant on.

chuckwagoncomputing, 2017-01-21:
What do I need to do to make the output constant on? With RgbColor(255,255,255) the outputs switch off for small amounts of time. RgbColor(256,256,256) gives a warning on compile: large integer implicitly truncated to unsigned type; and never switches the output on at all.

Makuna, 2017-01-22:
The gitter chat channel is a better place to ask this question as its not a real issue; yet.

[image]

255 is the max value ([see Wiki](https://github.com/Makuna/NeoPixelBus/wiki/RgbColor-object-API#properties)).
`RgbColor(255);` or the more verbose `RgbColor(255,255,255);` both create a full bright white color.  If you are correctly setting the color and calling `Show()` then it should stay solid and not flicker.

What hardware platform are you using?
How are you powering your NeoPixels (if by power supply, provide voltage and amp rating)?  
How many NeoPixels?
Do you have a sample sketch that demonstrates the output switching off?

penfold42, 2017-01-22:
How smalll is small ?

https://wp.josh.com/2014/05/15/getting-physical-to-uncover-neopixel-pwm-secrets/

We can see that the LED is never continuously on. Even when the color is set to full brightness, the LED still turns off for about 100us at the end of each PWM cycle

chuckwagoncomputing, 2017-01-22:
Ugh... That's exactly what's happening.

closed 2017-01-22 by chuckwagoncomputing