Makuna/NeoPixelBus #160: Support multiple output

Atraii, 2017-02-01:
NeoPixelBus is an awesome library, and it is currently possible to create two different strip outputs using two different methods. However,  this makes it confusing to update the strip. Consider an example scenario where you have 900 WS2812 LEDs. This many LEDs will have serious framerate issues (~35 fps max), but this can be doubled if the ESP8266 were to be placed in the middle of the strip and using two outputs. Then only 450 LEDs need to be updated with each data line. While NeoPixelBus can work with creating two different strip objects, it would be really great if only one 'multiple output' strip object had to be made, with the appropriate number of LEDs on each output specified.

I would really love to see a new method that combines DMA and the Async UART methods to supply one long continuous strip with the ESP8266 in the middle to output to both sides. Then internal functions could simply use `SetPixelColor` and it would adjust the DMA or UART pixels accordingly.

Makuna, 2017-02-01:
lets say both strips (A & B) have 4 pixels each.
Where does index 0 start?
```
  0   1   2   3   4   5   6   7 - index to combined strips
 A0  A1  A2  A3  B0  B1  B2  B3 - linear from first strip to second
 A3  A2  A1  A0  B0  B1  B2  B3 - linear from center out from each strip
```

Atraii, 2017-02-01:
I currently have the ESP module in the middle with one of the strip's reversed, like the second example you put. It's the easiest setup to connect. 

``` 
0   1   2    3   4   5   6   7 - index to combined strips 
A3 A2 A1 A0 B0 B1 B2 B3 - linear from center out from each strip 
```

peacepenguin, 2020-09-01:
https://github.com/Aircoookie/WLED/compare/master...peacepenguin:master

take a look at the fork differences in the repo i've created from WLED. WLED uses a "big single strip" throughout the entire program, similar to what you're after in concept. Then to enable multi-pin, i've just done the math on the provided "pixelindex" which represents the position on the giant "virtual" strip, the math just figures out what pixelindex to actually set on a specific strip.

Obviously you have to define the exact strip layout in advance so the math works. But the theory you're after is proven to be doable already without any changes or enhancements needed to neopixelbus.

Also check out the NeoBuffer method, you can generate a big "virtual" strip that way too, then dump the NeoBuffer to real strips using "Blt". Blt allows you to select an exact range of pixels to dump to the physical strip. So you can use the NeoBuffer as your virtual strip to begin, then at the end apply it to real NeoPixelBus strips using Blt.

@Makuna was a huge help on the gitter chat to help me figure all this out, and both methods work great. But seems to be an under-utilized, somewhat advanced feature of NeoPixelBus.

jvanlangen, 2020-10-18:
So is multiple ledstrips already fixed? (it's about 3 years ago now)

peacepenguin, 2020-10-18:
> So is multiple ledstrips already fixed? (it's about 3 years ago now) Still getting the RMT error. :-(

Yes.

Heres my test example for multtipin using blt method of neopixelbus:

https://www.github.com/peacepenguin/npb-complexblt/tree/master/src%2Fmain.cpp

You can send the same data to different strips, or unique data to each strip.

And here's a different way to do it, wled sends all output using neopixelbus, I've created a fork of wled that supports multipin output using neopixelbus, documented here:

https://github.com/Aircoookie/WLED/issues/104

Read from the bottom up, and find the most recent build instructions.

So yes I think it's proven to be possible now in two different ways using neopixelbus.

Use the link above in my previous post to my fork of wled, just click compare to see the changes I've made, all of it is reusable in other projects that use neopixelbus.

I would start with the fork of wled and get all your bugs worked out on code that has been tested successfully by a few people now, then take what you need from the wled fork.

jvanlangen, 2020-10-19:
Thanks for your reaction. I had to download a .zip file to add to my arduino IDE, is there a zip file with this multi-pin support also to download?

Regards,
Jeroen.

https://www.github.com/peacepenguin/npb-complexblt/tree/master/src%2Fmain.cpp

Aircoookie/WLED#104<https://github.com/Aircoookie/WLED/issues/104>

jvanlangen, 2020-10-19:
I checked your example, I'm doing the exact same. Still getting the "14:29:57.909 -> E (33) rmt: rmt_set_pin(397): RMT GPIO ERROR
" when I call strip2.begin();

jvanlangen, 2020-10-19:
Hmm, instead of using pin35, I used pin 22 (something to do with that?)

Makuna, 2020-10-19:
@jvanlangen Please don't use an issue to discuss your specific problem.  Please use the gitter channel.

Your question is different than this issue.  This issue is about a single instance of NeoPixelBus supporting multiple methods and pins; not multiple instances of NeoPixelBus (which has been working for years).

jvanlangen, 2020-10-19:
Ok,

I'm not really familiar with gitter channels of something like that yet.

Thanks for pointing that out.

Makuna, 2021-01-24:
https://github.com/Makuna/NeoPixelBus/pull/411
While this doesn't specifically fix this issue yet, it is a step toward it.

Makuna, 2023-02-12:
I keep thinking about this.  Especially now that we have the x8 and x16 i2s parallel methods where the multiple buses are used to improve speed.
Support for linear (single row) verses matrix (row/column) of pixel reference on input being mapped to mulitple buses.

Johboh, 2023-05-02:
As per https://github.com/Makuna/NeoPixelBus/discussions/694, I would love to get support for multiple strips together with the `NeoTiles`.
In my case, I have a panel of 7x5 tiles. Each tile is connected as `RowMajorAlternatingLayout`. Each column in the panel (a total of seven column) is a separate input/"strip", where five tiles in that column is interconnected using `RowMajorAlternatingLayout` (snakes all the way down).

Makuna, 2023-07-09:
Just the initial naming brain thought...
NeoMultiBus - an exposed single index range for multiple NeoPixelBus.  Usually for speed reasons or mixing different types of strip protocols together as one.  
NeoClusterBus - an exposed multi-index range (abstraction) for a single NeoPixelBus (and mulitple?).  Usually for treating a single strip as different entities to run different effects on or to improve readability of said effects.  The abstractions may infer a physical aspect that can be enumerated (think physical x,y or segment index) for input to the effect.