Makuna/NeoPixelBus #161: Compilation Failure: MKR1000, NeoPixelBus 2.2.6, Arduino 1.8.1

pret79, 2017-02-16:
I'm unable to compile the NeoPixelTest.ino code from the NeoPixelBus 2.2.6 examples for an Arduino/Genuino MRK1000 Board. It fails with the following compilation errors:

```
Arduino: 1.8.1 (Windows 10), Board: "Arduino/Genuino MKR1000"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Markus\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Markus\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Markus\Documents\Arduino\libraries -fqbn=arduino:samd:mkr1000 -ide-version=10801 -build-path C:\Users\Markus\AppData\Local\Temp\arduino_build_194023 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino5-static -prefs=runtime.tools.CMSIS.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Markus\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Markus\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Markus\Documents\Arduino\libraries -fqbn=arduino:samd:mkr1000 -ide-version=10801 -build-path C:\Users\Markus\AppData\Local\Temp\arduino_build_194023 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.9.0-arduino5-static -prefs=runtime.tools.CMSIS.path=C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0 -verbose C:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\examples\NeoPixelTest\NeoPixelTest.ino
Using board 'mkr1000' from platform in folder: C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12
Using core 'arduino' from platform in folder: C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12
Detecting libraries used...
"C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_SAMD_MKR1000 -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKR1000"' "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\cores\arduino" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\variants\mkr1000" "C:\Users\Markus\AppData\Local\Temp\arduino_build_194023\sketch\NeoPixelTest.ino.cpp" -o "nul"
"C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_SAMD_MKR1000 -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKR1000"' "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\cores\arduino" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\variants\mkr1000" "-IC:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "C:\Users\Markus\AppData\Local\Temp\arduino_build_194023\sketch\NeoPixelTest.ino.cpp" -o "nul"
"C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_SAMD_MKR1000 -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKR1000"' "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\cores\arduino" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\variants\mkr1000" "-IC:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\libraries\SPI" "C:\Users\Markus\AppData\Local\Temp\arduino_build_194023\sketch\NeoPixelTest.ino.cpp" -o "nul"
"C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_SAMD_MKR1000 -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKR1000"' "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\cores\arduino" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\variants\mkr1000" "-IC:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\libraries\SPI" "C:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HsbColor.cpp" -o "nul"
"C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m0plus -mthumb -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions  -w -x c++ -E -CC -DF_CPU=48000000L -DARDUINO=10801 -DARDUINO_SAMD_MKR1000 -DARDUINO_ARCH_SAMD  -D__SAMD21G18A__ -DUSB_VID=0x2341 -DUSB_PID=0x804e -DUSBCON '-DUSB_MANUFACTURER="Arduino LLC"' '-DUSB_PRODUCT="Arduino MKR1000"' "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS\4.5.0/CMSIS/Include/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\tools\CMSIS-Atmel\1.1.0/CMSIS/Device/ATMEL/" "-IC:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1. [... cut ...]
Bibliothek NeoPixelBus_by_Makuna in Version 2.2.6 im Ordner: C:\Users\Markus\Documents\Arduino\libraries\NeoPixelBus_by_Makuna  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Markus\AppData\Local\Arduino15\packages\arduino\hardware\samd\1.6.12\libraries\SPI  wird verwendet
exit status 1
```

Compiling the Sketch for other boards like an Arduino Uno works fine so that I assume it's an board-specific issue. The same error also happens for the Arduino Zero board.

Does the NeoPixelBus library support the MKR1000/Zero SAMD21 architecture?

penfold42, 2017-02-16:
I don't think those boards support the progmem directives so the library would need to be enhanced to not use progmem directives

Makuna, 2017-02-16:
This platform architecture is supported; but they may have changed/fixed things as Arm boards didn't correctly use Arduino compile define standards so there has to be extra checks than just the normal platform arch checks that are sort of hacks.
 
I do not see __ARM__ defined in the compile lines above; could you confirm in a sketch that it is present with your current build setup.  Add this line at the very top of your sketch and compile.
```
#if !defined(__arm__)
#error "Platform ARM NOT defined!"
#endif
```
Do you see the compile error in the output that states "Platform ARM NOT defined!"?

Makuna, 2017-02-16:
Problem found.
You should add an Issue with the ARM board project.  They removed the PGM_VOID_P type in some update.  

A work around is add the following to your sketch just before the include for my library...
```
// add the following for Arm as a work around until they fix their board support
#include <Arduino.h>
#ifndef PGM_VOID_P
#define PGM_VOID_P const void *
#endif
// add the above before the include for NeoPixelBus
#include <NeoPixelBus.h>
```

Makuna, 2017-06-18:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.2.8

closed 2017-06-18 by Makuna

foxcanada, 2017-11-11:
Any chance someone has this library running on rfduino?
What an awesome library! Thank you.  I was able to run this on a 300 led rgbw sk6812 strip.
However I'm a little too green to figure out how to compile on an rfduino board.
Could anyone provide some info? Need some info from the compile?

Makuna, 2017-11-11:
@foxcanada It not considered correct to reply to a closed issue unless it is directly related to said issue.  Since you mentioned another board than what this issue was created for, please don't do this.

You should look at the readme which states to go to gitter for quick questions (like yours).