Makuna/NeoPixelBus #162: Wdt reset cause 4 on ESP8266-01 and NodeMCU1.0

nitrique, 2017-02-23:
Hi guys,

I have a problem with your library, each time I add it to a sketch, it fail the wdt and reset like that : 

`wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld`

I've tested it with simple version 01 or 12E / Nodemcu and I've go the same problem.

I tried different programming speed and nodemcu v2 patch but nothing work.

Do someone know what to do ?

Thanks,
Nicolas

Makuna, 2017-02-23:
1) Have you tried one of the simple samples and confirmed they work?
2) Are you using the Serial object?  If so see issue https://github.com/Makuna/NeoPixelBus/issues/119 and read the wiki entry [NeoEsp8266Dma800KbpsMethod](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neoesp8266dma800kbpsmethod) which has the detail about serial.

if the samples work and you aren't using serial, then you need turn on debug features and get the stack (do a web search for "Arduino Esp8266 WDT crash stack"; it will entail sketch change and more.  This will tell you where in your sketch the problem is happening.

nitrique, 2017-02-24:
Hi Makuna,

First thanks for your work, it's ok with sk6812-rgbw :) !

It works now, I can't explain, it's weird, but I found that when it is on a breadboard, it cause reset cause 0 or 4 (I think about contact vibration due to high frequency).

I'll test with crash stack debugger, I didn't know this existed before, thanks.

I've seen a routine for that : 

1. sketch boot -> wifi say connecting to XXXXX -> and then the dots stop and wdt reset
2. sometimes it is ok without reboot.

I hope it will not occur when I'll make boards with that chip !

Makuna, 2017-02-24:
[NeoPixelBus Supported Pixels](https://github.com/Makuna/NeoPixelBus/wiki#supported-pixels)  ;-)

closed 2017-04-05 by Makuna