Makuna/NeoPixelBus #163: TM1829 Support

mattncsu, 2017-03-12:
Is it possible to adapt the code to add TM1829 support? I didn't even realize I had a TM1829 strip (sold as WS2811) until my neopixelbus program wasn't running the same.  Timing can't be too far off because they are lighting up using the NeoPixelFunRandomChange demo but no fade.

https://www.google.com/search?q=tm1829

Makuna, 2017-03-12:
Its not so easy.  
This is the [active issue with FastLED](https://github.com/FastLED/FastLED/issues/8).

What Arduino hardware are you using?
I don't own any to do any testing; and seems the timing is fiddly enough to require some.

Leaving active of course if I can find the time.

mattncsu, 2017-03-13:
I'm using an ESP8266 12E (NodeMCU). I saw there was a hacked Arduino-NeoPixel library ( http://www.tns-labs.org/tag/tm1829/) that supported the chip but I couldn't get any code with it to compile. It also seems the Spark/particle.io version of the library supported the TM1829. (https://gist.github.com/technobly/016690950d3b89cd2e20). Wasn't sure if it was possible to port that code over to arduino. Thanks for looking in to it!

Makuna, 2017-03-13:
At least with an Esp, the 1600mhz data rate might be possible.  But for AVR based Arduinos, the best they could do is the 800mhz data rate.

Makuna, 2021-03-21:
2.6.2 has this now.

closed 2021-03-21 by Makuna