Makuna/NeoPixelBus #164: wrong COPYING file

ensonic, 2017-03-14:
the sources state that the library is LGPL but the COPYING file is GPL. This is probably because on the GNU homepage the link to the LGPL is wrong (I've contected them).Here is the right one
https://www.gnu.org/licenses/lgpl.txt

Makuna, 2017-03-14:
thanks for catching this, I will get this corrected.

3rwww1, 2018-01-14:
Hi, any news about that ? We are about to release some code that links against NeoPixelBux, and we'd prefer the licence to be something else than GPL3 - probably MIT or even LGPL.

So how should we proceed ? Should we consider NeoPixelBus as GPL3 and release our own software as GPL3, or should we consider it LGPL3 so we can move on and release our own software as MIT (or something like that) ?

Thanks @Makuna !

Makuna, 2018-01-22:
Did you review the Adafruit library and what its licensing is? While not a direct fork from their code, there are lower level parts taken from their code base.

Makuna, 2018-01-22:
https://github.com/Makuna/NeoPixelBus/pull/196

closed 2018-01-22 by Makuna