Makuna/NeoPixelBus #165: Please note direct info about esp8266 pin

truebucha, 2017-04-01:
I searched info about the default pin looking through all wiki and found no mention. Only comment in test ino helped me well.
please add this
// For Esp8266, the Pin is omitted and it uses GPIO3 due to DMA hardware use.  
here 
[wiki](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object)

penfold42, 2017-04-01:
I agree it's not obvious but it is there:

https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neoesp8266dma800kbpsmethod

Makuna, 2017-04-02:
@truebucha, in the link you listed, its states it also.

"For the Esp8266 platforms, just omit the pin argument. This is due to hardware support that doesn't allow for selecting a pin."

closed 2017-06-18 by Makuna