Makuna/NeoPixelBus #166: fix AnimationParam progress value is rounded to 0.01

truebucha, 2017-04-02:
Possible I am doing something wrong, but animatior object gives me a values in range 0..1 but with precision 0.01 which not enough for me. (thinking in 0..255 color gradient range it should be at least 255 steps between 0..1 in progress, better 1000)

So please make precision of it at least 0.001 or provide a parameter to able set up.

thanks a lot

Makuna, 2017-04-02:
This is a better question to ask the Gitter channel than a real issue/bug.  Please see the readme for the link to gitter channel.

Just to be clear, are you stating you are getting a range between 0.0 and 0.1?  Or between 0.0 and 1.0?  Its hard to read it how you entered.

The value should be between 0.0 and 1.0.  Thus, a precision of 0.002 is 500 steps.  0.2 would be 5 steps.  0.02 would be 50 steps.

The precision is directly related to how often you call the `animation.UpdateAnimations()` in your main loop AND how much time you take to do other actions in your main loop.

Makuna, 2017-04-02:
How long is your animation?
How much time do you take in your loop?  You can capture this by sampling microseconds() at the top and bottom and print them out; capture the min and max and a running average will also help diagnose issues.

truebucha, 2017-04-03:
Thank you for reply.
I tried all types of animation time argument provided in the animator when creating instance. I use default loop on esp8266 without any delay. The main misunderstanding is why when I use seconds for animation time I get progress 0.01, 0.02, 0.03 and when I use milliseconds I get 0.002, 0.004, 0.006 with the same loop count for 5 sec animation

Makuna, 2017-04-03:
Please be more specific in your statements, provide a code snippet that demonstrates it.

You state, "when I use seconds for animation time", do you mean you construct the animation object defined to use seconds for the time scale like the following?

```
```

truebucha, 2017-04-06:
I mean actually that you described. But why it should force the FLOAT progress indicator values be different for NEO_SECONDS and NEO_MILLISECONDS on the same 5 sec animation with the same loop function calls count?

truebucha, 2017-04-06:
`#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

typedef NeoGrbFeature MyPixelColorFeature;

// use this for millisec
//uint16_t timeInterval = NEO_MILLISECONDS;
//int animationTime = 5;  

// use this for sec
uint16_t timeInterval = NEO_SECONDS;
uint16_t animationTime = 5; 

uint16_t loopsCount = 0;

const uint16_t PixelCount = 144; // the sample images are meant for 16 pixels 

NeoPixelBus<MyPixelColorFeature, Neo800KbpsMethod> strip(PixelCount);
NeoPixelAnimator animations(1, timeInterval); // NeoPixel animation management object

void updateAnimation(AnimationParam param) {
  if (param.state == AnimationState_Completed) {
      Serial.println("Total loops:");
      Serial.println(loopsCount);
      return;
  }

   float progress = param.progress;
   Serial.println((progress * 10000.0f));
}

void setup() {
    Serial.begin(230400);
    strip.Begin();
    strip.Show();
    animations.StartAnimation(0, 100, updateAnimation);
}

void loop() {
    animations.UpdateAnimations();
    strip.Show();
    loopsCount += 1;
}`

Makuna, 2017-04-06:
I think you are miss understanding how you use the timer interval value.  By setting the value to seconds, it means you loose incremental resolution but gain longer animations.  It is meant to be used to increase the animation time period NOT to provide an easier value number range.  It has consequences the larger you make it..

From the sample....
```
// Possible values from 1 to 32768, and there some helpful constants defined as...
// NEO_MILLISECONDS        1    // ~65 seconds max duration, ms updates
// NEO_CENTISECONDS       10    // ~10.9 minutes max duration, centisecond updates
// NEO_DECISECONDS       100    // ~1.8 hours max duration, decisecond updates
// NEO_SECONDS          1000    // ~18.2 hours max duration, second updates
// NEO_DECASECONDS     10000    // ~7.5 days, 10 second updates
```
If you set it to seconds, you now can have animations that run for over 18 hours; but the resolution of changes is about a second.  So any loop calls faster than a second will provide near the same progress value.
Most people are happy with the default as it provides possible animations length of just over 65 seconds.  If your animations need to run longer, you need to sacrifice resolution to gain the longer timer period.

truebucha, 2017-04-10:
You are right

closed 2017-04-10 by truebucha