Makuna/NeoPixelBus #168: Optimisation for WS2813

superjyce, 2017-04-07:
Hi,
The WS2813 led is an improvement from the WS2812.
It support the ws2812 protocol but timings change a bit.
For exemple : 
- RST minimum time frame increase to 300us
- TH+TL=1.1µs±300ns

Is it possible to add the optimized support of this led on the library ?

Datasheet : http://www.szledcolor.com/download/WS2813%20LED.pdf

Makuna, 2017-04-07:
The timing is very short. Some of the AVRs might not be able to achieve that.

What hardware platform do you use?

superjyce, 2017-04-07:
Thanks for your answer.
I'm on an ESP8266 using UART method (because of NodeMCU UART0 is already taken by USB chip).

I tweaked your library to adjust the baudrate (to 3636363 which gave me a low level value of 22 with a 80Mhz bus clock), the time to send a pixel (to 27us) and the RES time (from 50 to 300). 

Now I need some time to make tests.

Makuna, 2017-04-08:
For the ESP this is much easier to accomplish; but I try not to have a feature that is for a specific platform only.

superjyce, 2017-04-08:
I anderstand your point.
But I think its better to have something for some plateform than nothing.

penfold42, 2017-04-08:
Just been comparing data sheets

If I'm reading it right the timing is no shorter than the 800kHz 2812 protocol.

The biggest difference is the much longer reset pulse

Makuna, 2017-04-08:
@penfold42 You are correct, I was reading it incorrectly.  Its strange they state a range that the first value is us and the second is ns.  The tolerance on the total bit length is tighter (300ns versus the normal 600ns).

Then this should be pretty easy.  

Neo800KbpsMethod will probably have an alternative, NeoWs2813Method.

superjyce, 2017-04-08:
@Makuna, it should be OK with this :
`    bool IsReadyToUpdate() const
    {
        uint32_t delta = micros() - this->_startTime;
        return delta >= getPixelTime() + 300; // Intead of 50 for WS2813
    }` 
`class MyNeoEsp8266UartSpeed800Kbps
{
public:
  static const uint32_t ByteSendTimeUs = 27; // us it takes to send a single pixel for WS2813
  static const uint32_t UartBaud = 3636363; // ((1 / 1.1us) * 4 * 1 000 000).
};
typedef NeoEsp8266UartMethodBase<MyNeoEsp8266UartSpeed800Kbps, NeoEsp8266AsyncUart> MyNeoEsp8266AsyncUart800KbpsMethod;
`
For the WS2812, you set ByteSendTimeUs = 10; but it should be 30 isn't it ?

Makuna, 2017-04-08:
@superjyce Have you tried it without a change and a simple `delayMicroseconds(300)' right after a `strup.show()` and confirmed that it works for you?

ByteSendTimeUs should be 1.25 * 8; which is 10.  A single bit should be 1.25us +- 0.6us, bits per byte.  This isn't timer per pixel, as depending on the number of bytes per pixel (some are 4 bytes per pixel like RGBW).

Makuna, 2017-04-09:
https://github.com/Makuna/NeoPixelBus/pull/170
This has been merged, but an official build has not been published until more testing can be done.
I would appreciate that you manually install the current github files and test it with the actual Ws2813 legs as I do not have any and all my testing will just be with a digital logic probe.

superjyce, 2017-04-09:
Thanks.
I'll not have time this week but I'll test the next one.

superjyce, 2017-04-23:
Ok. Seems there is no difference but it works :)

closed 2017-04-23 by superjyce