Makuna/NeoPixelBus #172: STM32L4 support

Jorgen-VikingGod, 2017-06-06:
Is there a easy way to get also support for STM32L4 based MCUs?
I'm using the NUCLEO-L432 with following ArduinoIDE support:
https://github.com/GrumpyOldPizza/arduino-STM32L4

but I got some build errors, if I want to use NeoPixelBus:
```
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:582:103: error: field initializer is not constant
     static const uint32_t CyclesT0h = ((uint32_t)(0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:583:103: error: field initializer is not constant
     static const uint32_t CyclesT1h = ((uint32_t)(0.80 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:584:100: error: field initializer is not constant
     static const uint32_t Cycles = ((uint32_t)(1.25 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                    ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:591:103: error: field initializer is not constant
     static const uint32_t CyclesT0h = ((uint32_t)(0.40 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:592:103: error: field initializer is not constant
     static const uint32_t CyclesT1h = ((uint32_t)(0.80 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:593:100: error: field initializer is not constant
     static const uint32_t Cycles = ((uint32_t)(1.25 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                    ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:600:103: error: field initializer is not constant
     static const uint32_t CyclesT0h = ((uint32_t)(0.50 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:601:103: error: field initializer is not constant
     static const uint32_t CyclesT1h = ((uint32_t)(1.20 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:602:100: error: field initializer is not constant
     static const uint32_t Cycles = ((uint32_t)(2.50 * ARM_OTHER_SCALE + 0.5) - (5 * ARM_OTHER_INST));
                                                                                                    ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h: In static member function 'static void NeoArmOtherSpeedBase<T_SPEEDPROPS>::send_pixels(uint8_t*, size_t, uint8_t)':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:615:9: error: 'Pio' was not declared in this scope
         Pio* port;
         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:615:14: error: 'port' was not declared in this scope
         Pio* port;
              ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:616:18: error: 'WoReg' does not name a type
         volatile WoReg* portSet;
                  ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:617:18: error: 'WoReg' does not name a type
         volatile WoReg* portClear;
                  ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:618:18: error: 'WoReg' does not name a type
         volatile WoReg* timeValue;
                  ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:619:18: error: 'WoReg' does not name a type
         volatile WoReg* timeReset;
                  ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:625:35: error: there are no arguments to 'pmc_set_writeprotect' that depend on a template parameter, so a declaration of 'pmc_set_writeprotect' must be available [-fpermissive]
         pmc_set_writeprotect(false);
                                   ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:625:35: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:626:41: error: 'TC3_IRQn' was not declared in this scope
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
                                         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:626:49: error: there are no arguments to 'pmc_enable_periph_clk' that depend on a template parameter, so a declaration of 'pmc_enable_periph_clk' must be available [-fpermissive]
         pmc_enable_periph_clk((uint32_t)TC3_IRQn);
                                                 ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:628:22: error: 'TC1' was not declared in this scope
         TC_Configure(TC1, 0,
                      ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:629:13: error: 'TC_CMR_WAVE' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
             ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:629:27: error: 'TC_CMR_WAVSEL_UP' was not declared in this scope
             TC_CMR_WAVE | TC_CMR_WAVSEL_UP | TC_CMR_TCCLKS_TIMER_CLOCK1);
[... 15 lines cut ...]
                                       ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:634:9: error: 'portSet' was not declared in this scope
         portSet = &(port->PIO_SODR);            // burn a few cycles after
         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:635:9: error: 'portClear' was not declared in this scope
         portClear = &(port->PIO_CODR);            // starting timer to minimize
         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:636:9: error: 'timeValue' was not declared in this scope
         timeValue = &(TC1->TC_CHANNEL[0].TC_CV);  // the initial 'while'.
         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:637:9: error: 'timeReset' was not declared in this scope
         timeReset = &(TC1->TC_CHANNEL[0].TC_CCR);
         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:658:26: error: 'TC_CCR_CLKEN' was not declared in this scope
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                          ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:658:41: error: 'TC_CCR_SWTRG' was not declared in this scope
             *timeReset = TC_CCR_CLKEN | TC_CCR_SWTRG;
                                         ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:678:23: error: there are no arguments to 'TC_Stop' that depend on a template parameter, so a declaration of 'TC_Stop' must be available [-fpermissive]
         TC_Stop(TC1, 0);
                       ^
exit status 1
Error compiling for board NUCLEO-L432KC.
```

Makuna, 2017-06-06:
This library currently doesn't support it.  I don't have one to even test for it.
The code path is using a generic ARM, which needs VARIANT_MCK to be defined; which the board you are using doesn't seem to define this.
Do you know if Adafruits library supports it?

Jorgen-VikingGod, 2017-06-06:
sorry also issues on building: https://github.com/adafruit/Adafruit_NeoPixel/issues/140

Makuna, 2017-06-06:
before you include NeoPixelBus, add
```
#define ARDUINO_STM32_FEATHER
```
If you turn on verbose within the IDE, and include a few lines of the compile before the first error (you will see lines that include the defined like "-Dblahblah -Dblahbhal" that would give me the defined for the specific board you are compiling for.

Jorgen-VikingGod, 2017-06-07:
with the define I only got one error left. PIN_MAP is missing:

```
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h: In static member function 'static void NeoArmStm32SpeedBase<T_SPEEDPROPS>::send_pixels(uint8_t*, size_t, uint8_t)':
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:524:33: error: 'PIN_MAP' was not declared in this scope
         uint32_t  pinMask = BIT(PIN_MAP[pin].gpio_bit);
                                 ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:524:54: error: there are no arguments to 'BIT' that depend on a template parameter, so a declaration of 'BIT' must be available [-fpermissive]
         uint32_t  pinMask = BIT(PIN_MAP[pin].gpio_bit);
                                                      ^
/Applications/Arduino-1.8.0.app/Contents/Java/libraries/NeoPixelBus/src/internal/NeoArmMethod.h:524:54: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
exit status 1
Error compiling for board NUCLEO-L432KC.
```

Jorgen-VikingGod, 2017-06-07:
I have a dirty solution:
just change this lines:
https://github.com/Makuna/NeoPixelBus/blob/master/src/internal/NeoArmMethod.h#L524-L527
```
        uint32_t  pinMask = BIT(PIN_MAP[pin].gpio_bit);

        volatile uint16_t* set = &(PIN_MAP[pin].gpio_device->regs->BSRRL);
        volatile uint16_t* clr = &(PIN_MAP[pin].gpio_device->regs->BSRRH);
```
into this:

```
        uint32_t pinMask = g_APinDescription[pin].bit;
        GPIO_TypeDef *GPIO = g_APinDescription[pin].GPIO;

        volatile uint16_t* set = &(GPIO->BRR);
        volatile uint16_t* clr = &(GPIO->BSRR);
```

Makuna, 2019-06-15:
BTW, what I was asking for I think I found elsewhere, the arch define is `ARDUINO_ARCH_STM32F4`, I hope this is what yours was.

Makuna, 2019-06-15:
https://github.com/Makuna/NeoPixelBus/pull/274

Makuna, 2019-06-15:
And BTW, some of these DO NOT provide the strcpy_P that most boards support.  So if you run into this, create an issue with the board support specifics to your hardware.

Makuna, 2019-09-02:
2.5.1

closed 2019-09-02 by Makuna