Makuna/NeoPixelBus #174: Invert signal for esp8266

denruss, 2017-08-06:
I have a little idea. Make the setting in NeoPixelBus so that the control signal is inverse. This is very convenient for ESP8266, it would be possible to assemble the 3.3 to 5v level converter on one n-channel transistor.

Invert the signal is fairly simple, you need to file NeoEsp8266DmaMethod.h
Invert all bits of this structure
        Const uint16_t bitpatterns [16] =
        {
            0b1000100010001000, 0b1000100010001110, 0b1000100011101000, 0b1000100011101110,
            0b1000111010001000, 0b1000111010001110, 0b1000111011101000, 0b1000111011101110,
            0b1110100010001000, 0b1110100010001110, 0b1110100011101000, 0b1110100011101110,
            0b1110111010001000, 0b1110111010001110, 0b1110111011101000, 0b1110111011101110,
        };};

And replace memset (_i2sZeroes, 0x00, sizeof (_i2sZeroes));
On memset (_i2sZeroes, 0xFF, sizeof (_i2sZeroes));

sorry for my English

gitolicious, 2018-10-09:
I like the idea, though I need it for bit bang method, not DMA method. Any clue where I need to change something? My guess is somewhere along these lines:
https://github.com/Makuna/NeoPixelBus/blob/6ca73195fb5178bc2cff53e9bf595e861db7c99d/src/internal/NeoEspBitBangMethod.h#L47
https://github.com/Makuna/NeoPixelBus/blob/6ca73195fb5178bc2cff53e9bf595e861db7c99d/src/internal/NeoPixelEsp.c#L68

Makuna, 2018-10-10:
@gitolicious Are you stating you hare using Bitbang on Esp8266?  Its not going to be very stable if you are using WiFi at all.

I am still working on how to expose it in a general way.

gitolicious, 2018-10-10:
Thanks for the hint, I have read about it in the wiki. In fact it's not my own code...
https://github.com/arendst/Sonoff-Tasmota/blob/ea43540b436de459118b70214f4e039c43b794ee/sonoff/xplg_ws2812.ino#L44
Not seeing any troubles so far though. For that reason I wanted to give it a try. Trying to avoid DMA mode as GPIO3 is not easily available on my board (Arilux LC10).
Any ideas how to continue?

Makuna, 2018-10-10:
@gitolicious BTW, there is also the UART method, locked to GPIO2.

To modify the big bang is probably just a few places; for your own use of course ;-)

NeoEspBitBangMethod..h(111) Change LOW to HIGH
NeoPixelEsp.c(80 & 93) swap the lines at each locations with the other location 

That should work.

gitolicious, 2018-10-10:
Perfect, will give it a try and report back.
Sure it's just for my personal code in this specific project and use case.

gitolicious, 2018-10-11:
Looks like that worked out perfectly on the software side of things! Unfortunately now I am struggling on the hardware side. Need to dig out my soldering iron which I tried to avoid. Thanks a lot @Makuna for the immediate response!

Makuna, 2018-10-11:
I wish it was universally that easy in all the methods.

zanglerecke, 2019-09-26:
I want to use also a inverting circuit,... (I am not only inverting, i just use an inverting optocoupler for galvanically seperating the VCC of the ESP32 and the one of the led-Strap. 

for that, one could be use an additional class in NeoEsp32RmtMethod.h

just copy line 91 to 97 and swap the timings of 0 and 1 (400,850) <-> (800,450)
also make a new NeoEsp32Rmt0800KbpsMethodINV with that option

so far this adaptions are working BUT:

my first LED is not working at all (but it is not defect).
maybe there is no accurate reset signal? 
how is the reset delay realized? (is it the value near line 300?)

Makuna, 2019-09-27:
@zanglerecke Look at the following line to invert the reset, change from LOW to HIGH...
```
config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
```
Then, instead of what you trying to do, change this method which will invert the data pulses
```
    inline constexpr static uint32_t Item32Val(uint16_t nsHigh, uint16_t nsLow)
    {
        return (FromNs(nsLow) << 16) | (1 << 15) | (FromNs(nsHigh));
    }
```
to 
```
    inline constexpr static uint32_t Item32Val(uint16_t nsHigh, uint16_t nsLow)
    {
        return (FromNs(nsLow) << 16) | (1 << 31) | (FromNs(nsHigh));
    }
```

Makuna, 2019-09-27:
I have confirmed the changes above work for Esp32 RMT method. 
I am working on the default Esp32 * Esp8266 methods along with bitbang for both, and then I will publish it.  Probably also Esp8266 Uart method.

zanglerecke, 2019-09-27:
Thank you, 
your proposals where 100% correct. But this modification will change also every other setup. (for this time, i can live with that.

works like a charm

Makuna, 2019-09-28:
https://github.com/Makuna/NeoPixelBus/pull/296

most of the esp8266 and esp32 methods now have a inverted version, just add the word "Inverted" before the "Method", like

NeoEsp8266BitBang800KbpsMethod => NeoEsp8266BitBang800KbpsInvertedMethod