Makuna/NeoPixelBus #175: Problem building the project using Platformio and ESP32

phenzao, 2017-08-15:
I am trying to build the NeoPixelTest using Platformio but I get this error: "undefined reference to `bitbang_send_pixels_800'.  I tried to build the project with arduino IDE and it is working correctly. Any one have any idea what I need to do for the Platformio to work?

Makuna, 2017-08-15:
I don't specifically support PlatformIO issues.

But, look through the verbose compile output and look for warnings/errors.  Also grab any relevant lines that reference the library and include those.  Maybe someone else with more knowledge with PlatformIO can help.

EBZ-Krisemendt, 2017-09-06:
i experienced several problems like this with external libraries using platformio.
started to manually clone those in each projects lib folder. so far all those "undefined reference"/"not found"/etc errors seemed to be fixed by this procedure.

Makuna, 2017-09-06:
A shot in the dark, but  check if the includes if they use double quote (") or greater/less than (<>) around the file name; quote is supposed to be local files, greater than equal is supposed to be external libraries; Arduino this isn't often an issue so you see incorrect use of them out there in libraries.

hallard, 2018-03-27:
I've got the same problem, copying library in local lid does not solve it.

Here the full log output if it can help

```
Compiling .pioenvs\lolin32_lora\src\blecount.cpp.o
Compiling .pioenvs\lolin32_lora\src\configmanager.cpp.o
Compiling .pioenvs\lolin32_lora\src\lorawan.cpp.o
Compiling .pioenvs\lolin32_lora\src\main.cpp.o
pio : In file included from C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/WString.h:29:0,
Au caractère Ligne:1 : 1
+ pio run 1> file.log 2>&1
+ ~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (In file include...WString.h:29:0,:String) [], RemoteException
    + FullyQualifiedErrorId : NativeCommandError
 
from C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/Arduino.h:150,

from src\globals.h:2,

from src\lorawan.cpp:3:

.piolibdeps\NeoPixelBus_ID547\src/internal/HtmlColor.h: In member function 'size_t HtmlColor::Parse(const char*, size_t)':

C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/pgmspace.h:53:14: error: there are no arguments to 'typeof' that depend on a template 
parameter, so a declaration of 'typeof' must be available [-fpermissive]

typeof(addr) _addr = (addr); \

^

.piolibdeps\NeoPixelBus_ID547\src/internal/HtmlColor.h:215:47: note: in expansion of macro 'pgm_read_ptr'

PGM_P searchName = (PGM_P)pgm_read_ptr(&colorPair->Name);

^

C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/pgmspace.h:53:14: note: (if you use '-fpermissive', G++ will accept your code, but allowing 
the use of an undeclared name is deprecated)

typeof(addr) _addr = (addr); \

^

.piolibdeps\NeoPixelBus_ID547\src/internal/HtmlColor.h:215:47: note: in expansion of macro 'pgm_read_ptr'

PGM_P searchName = (PGM_P)pgm_read_ptr(&colorPair->Name);

^

C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/pgmspace.h:53:16: error: expected ';' before '_addr'

typeof(addr) _addr = (addr); \

^

.piolibdeps\NeoPixelBus_ID547\src/internal/HtmlColor.h:215:47: note: in expansion of macro 'pgm_read_ptr'

PGM_P searchName = (PGM_P)pgm_read_ptr(&colorPair->Name);

^

C:\users\charles\.platformio\packages\framework-arduinoespressif32\cores\esp32/pgmspace.h:54:21: error: '_addr' was not declared in this scope

*(void * const *)(_addr); \

^

.piolibdeps\NeoPixelBus_ID547\src/internal/HtmlColor.h:215:47: note: in expansion of macro 'pgm_read_ptr'

[... 555 lines cut ...]

.piolibdeps\NeoPixelBus_ID547\src/internal/NeoColorFeatures.h:136:22: error: void value not ignored as it ought to be
*pDest++ = pgm_read_dword(pSrc++);
^

*** [.pioenvs\lolin32_lora\src\configmanager.cpp.o] Error 1

*** [.pioenvs\lolin32_lora\src\lorawan.cpp.o] Error 1

*** [.pioenvs\lolin32_lora\src\main.cpp.o] Error 1

*** [.pioenvs\lolin32_lora\src\blecount.cpp.o] Error 1

 [ERROR] Took 5.86 seconds 

 [SUMMARY] 
Environment heltec_wifi_lora_32	[SKIP]
Environment ttgov1             	[SKIP]
Environment ttgov2             	[SKIP]
Environment lopy               	[SKIP]
Environment lopy4              	[SKIP]
Environment lolin32_lora       	[ERROR]

 [ERROR] Took 5.86 seconds 
```

nk36, 2018-05-02:
Hi,

I have exactly the same problem and it is a shame :(

Hope I will find an answer somewhere!

Makuna, 2018-05-02:
What version of the Esp32 Arduino are you using?

nk36, 2018-05-03:
I'm working on the heltec wifi lora 32, this model precisely :
https://www.banggood.com/fr/Wemos-TTGO-LORA-SX1278-ESP32-0_96OLED-16-Mt-Bytes-128-Mt-bit-433Mhz-For-Arduino-p-1205930.html?cur_warehouse=CN

Makuna, 2018-08-16:
A lot has changed with Esp32 (formal release has been made) and I have since added hardware support.  Is this issue still a problem?

Makuna, 2019-02-05:
closing due to lack of communications. 
I don't believe this issue really has anything to do with this library.

closed 2019-02-05 by Makuna

VioletGiraffe, 2019-11-03:
Sorry to bother you as this indeed is not a library issue per se, but having just spent 20 minutes figuring it out, I want to report my findings as this issue is among the first Google results on the topic.

Apparently, `typeof` is a GCC extension. As such, it is disabled when the program is compiled with `std=c++XX`. There are two ways to solve this:
1. Use `__typeof__` instead of `typeof` (can be done with a global define: `build_flags = -Dtypeof=__typeof__`
2. Use `std=gnu++XX`.

Makuna, 2019-11-03:
@VioletGiraffe The typeof is not in NeoPixelBus, it is in the Esp32 Arduino.

VioletGiraffe, 2019-11-03:
@Makuna: understood, removed the post scriptum. Thanks for the awesome library, by the way!

osmeest, 2023-12-18:
Seasoned C++ developer, I'm a bit disappointed by the quality of ESP32 libraries. They suck big time when it comes to C++. All those #define creating incompatibilities between libraries. As if namespaces, templates, ... were there for nothing.
@VioletGiraffe your last comment saved me from hours of torturing myself.
@Makuna, indeed, this has nothing to do with your lib which is very nice but the solution to my problem happened to be here :)