Makuna/NeoPixelBus #177: Does not compile

CrabbyPete, 2017-08-28:
My project compile fine with and esp12 dev board, but when I tried it with an esp32 dev board I got
Arduino: 1.6.7 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

Build options changed, rebuilding all
In file included from C:\Users\Pete\Documents\Arduino\libraries\NeoPixelBus\src/NeoPixelBus.h:92:0,

                 from C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus\src/NeoPixelBrightnessBus.h:29,

  
C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus\src/internal/DotStarSpiMethod.h:29:17: fatal error: SPI.h: No such file or directory

compilation terminated.

exit status 1
Error compiling.

Makuna, 2017-08-28:
Please update your Arduino IDE, 1.6.7 is pretty old and had many issues with libraries.

CrabbyPete, 2017-08-29:
I upgraded and got
In file included from sketch\patterns.cpp:7:0:

C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h: In instantiation of 'NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::NeoPixelBus(uint16_t) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEspBitBangMethodBase<NeoEspBitBangSpeed800Kbps>; uint16_t = short unsigned int]':

sketch\patterns.cpp:47:78:   required from here

C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:113:56: error: no matching function for call to 'NeoEspBitBangMethodBase<NeoEspBitBangSpeed800Kbps>::NeoEspBitBangMethodBase(uint16_t&, const size_t&)'

         _method(countPixels, T_COLOR_FEATURE::PixelSize)

                                                        ^

In file included from C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:68:0,

                 from sketch\patterns.cpp:7:

C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEspBitBangMethod.h:63:5: note: candidate: NeoEspBitBangMethodBase<T_SPEED>::NeoEspBitBangMethodBase(uint8_t, uint16_t, size_t) [with T_SPEED = NeoEspBitBangSpeed800Kbps; uint8_t = unsigned char; uint16_t = short unsigned int; size_t = unsigned int]

     NeoEspBitBangMethodBase(uint8_t pin, uint16_t pixelCount, size_t elementSize) :

     ^

C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEspBitBangMethod.h:63:5: note:   candidate expects 3 arguments, 2 provided

C:\Users\Peter\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src/internal/NeoEspBitBangMethod.h:60:34: note: candidate: constexpr NeoEspBitBangMethodBase<NeoEspBitBangSpeed800Kbps>::NeoEspBitBangMethodBase(const NeoEspBitBangMethodBase<NeoEspBitBangSpeed800Kbps>&)

 template<typename T_SPEED> class NeoEspBitBangMethodBase

From: Michael Miller
Sent: Monday, August 28, 2017 5:32 PM
To: Makuna/NeoPixelBus
Cc: Pete Douma; Author
Subject: Re: [Makuna/NeoPixelBus] Does not compile (#177)

Please update your Arduino IDE, 1.6.7 is pretty old and had many issues with libraries.

Makuna, 2017-08-29:
Neither I nor others are seeing this problem, so I suspect its some sort of setup/config issue.

Could you include the sketch that you are having a problem with?  Or better, a minimum sketch in size that still demonstrates the problem?  (if you have a blank line with elipses at the top and bottom of the code, it will get formatted as code in this editor.

Please check your IDE/Library Manager/ for the version of the NeoPixelBus library and include that in a response.  
Please check you IDE/Device Manager/ for the version of the Esp32 support and include that in a response.

How did you install the library?  From Github or through the library manager?

phil-hx, 2018-01-06:
I had the same problem, 

I just used   NeoEspBitBangSpeedWs2813  instead of NeoEsp32BitBangWs2813Method in the strip declaration.

If you use the correct type there is no problem.

Makuna, 2018-08-16:
I am confused as NeoEspBitBangSpeedWs2813 is not a valid method, is a speed feature class that is used to create the actual Method class.

For esp32, there is only three valid bit bang methods to use, 
NeoEsp32BitBangWs2813Method
NeoEsp32BitBang800KbpsMethod
NeoEsp32BitBang400KbpsMethod

Since this issue was opened, the default three no longer use bitbang and bitbang should be avoided.  The default three now use hardware and I suspect this issue is no longer valid.
These are...
NeoWs2813Method
Neo800KbpsMethod
Neo400KbpsMethod

closed 2018-08-16 by Makuna