Makuna/NeoPixelBus #178: DMA on Adafruit Huzzah 8266

CrabbyPete, 2017-08-31:
I am using the DMA on the Adafruit Huzzah 8266 [https://www.adafruit.com/product/2471?gclid=Cj0KCQjw557NBRC9ARIsAHJvVVO6l3QSObFHc0n4BozSEH1nhtn5ellP5d8p9GkKPi0hABYCQg1brlkaAuUYEALw_wcB](url) and it does not seem to work. It works using the UART.  Does this board interfere with the RX0 pin?

Makuna, 2017-08-31:
It states it has "Level shifting on the UART and reset pin,"  I suspect they are doing "one way" level shifting, since the default for RX is "IN" and the DMA remaps this to an "OUT", I suspect this is the issue.

Makuna, 2018-08-16:
confirmed this is an issue with this board.  Use the alternate UART method exposed for esp8266 to work around this boards issue with the default DMA method.

closed 2018-08-16 by Makuna