Makuna/NeoPixelBus #179: Confusing comment in examples

stoefln, 2017-09-04:
Maybe you can help me to understand the following comment, found in various examples of your library:
> const uint8_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266

I am following NodeMcu board (ESP8266): https://goo.gl/mBPXsR
And I am wondering which pin to use for my LED strip. 

I also read through the FAQ, about DMA and modes, but I guess I am not familiar enough with these technical details to understand. I also wonder why the Adafruit Neopixel library has a setting for LED pin which works on the ESP8266.

Can you help me figuring out the right pin configuration?

Makuna, 2017-09-07:
This is better asked on the gitter channel.

 See [FAQ #1](https://github.com/Makuna/NeoPixelBus/wiki/FAQ-%231) where it states ...
"Due to restrictions on the hardware used for these two methods there is no way to support other pins."
Then read [NeoEsp8266BitBang800KbpsMethod](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neoesp8266bitbang800kbpsmethod)

Most boards you can pick the pin you want to use.  With the Esp8266 you are stuck with using specific pins depending on the "method" you supply at your constructor.  

The BitBang method relies on strict timing with interrupts turned off.  But the WiFi interrupts CAN NOT be blocked without causing crashes.    Currently the AdaFruit library (which is based on my bitbang method) can not be used. 
Note that it may work for a bit, the crash is random due to timing if the WiFi tries to interrupt during the time the interrupts are turned off.

stoefln, 2017-09-13:
Thanks so much for your explanations- especially the random crashes would have caused me a lot of headache, if you wouldn't have told me!
Using NeoEsp8266Uart800KbpsMethod and GPIO4 (D2 on the NodeMcu) worked for me in the end.
NeoEsp8266Dma800KbpsMethod did not work, because the LEDs light up, and are causing a crash  during flashing.

closed 2017-09-13 by stoefln