Makuna/NeoPixelBus #180: Setting color on one LED causes others to change

freeheel3r, 2017-09-19:
Hi,

I'm not sure if this is a problem with the library or my LED strip (I only have one, so I can't test another to make sure), but occasionally when calling setColor on one LED, another or several other LEDs will also change color. This appears to be random, as different LEDs are switched each time.

I am using an ESP32, and initializing the strip with the Neo800kbps method, if that changes anything. The strip is 37 LEDs long.

Also, this happens whether I power the strip with 3.3V and connect data directly to the ESP32, or I power the strip with 5V and use a level shifter (BSS138) to shift the data line to 5V.

Any ideas as to what could be causing the issue?

Makuna, 2017-09-19:
I suspect I do.

See https://github.com/adafruit/Adafruit_NeoPixel/issues/139.

It turns out that even on the Esp32, bitbang can not be achieved even though the primary core should be available.

https://github.com/Makuna/NeoPixelBus/issues/152 is tracking adding the hardware support.

freeheel3r, 2017-09-19:
Changing to ws2813 method didn't help, but adding `delay(1)` before every call to `Show()` and cutting down to 32 LEDs (I didn't use the last 5 anyway) helped quite a lot.

There is still the occasional flickering / wrong color, but it is much more rare now. 

Thank you very much for your help, and your work developing this library!

Makuna, 2017-10-08:
A possible related issue I found https://github.com/Makuna/NeoPixelBus/issues/182
I am suspecting this maybe the true problem you are seeing with BitBang.
I will get an update out soon with this (week or two).

Makuna, 2017-10-22:
@freeheel3r Could you pick up the Github version and try it and see if it fixes your issue.

Makuna, 2018-08-16:
closed due to lack of response and believed to be fixed

closed 2018-08-16 by Makuna