Makuna/NeoPixelBus #184: Animations flicker like crazy

Fernando-Checa, 2017-10-12:
Hello,
I'm trying to do some basic animations, a fade in for example, and the strip flickers all over the place.
I'm using Esp8266 with ws2812b strips, Arduino 1.8.5, neopixelbus 2.2.8.
Also tried using ws2813 code with no success.
The program works fine with Arduino board, both 5v and 3.3v.
Tried level shifter and diode hack. Nothing works.

None of the examples work.

Any idea of what could be wrong? 
Any help would be appreciated

Here's a video showing the problem:
[https://youtu.be/_x1ZaQCpapU](url)

and code is pretty basic:

`
#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

// How many leds are in the strip?
#define NUM_LEDS 117

const uint16_t PixelCount = NUM_LEDS; // make sure to set this to the number of pixels in your strip
const uint16_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266
const uint16_t AnimCount = 1; // we only need one
const uint16_t TailLength = 6; // length of the tail, must be shorter than PixelCount

const float MaxLightness = 0.4f; // max lightness at the head of the tail (0.5f is full bright)

uint16_t panel_actual[NUM_LEDS];
uint16_t panel_dest[NUM_LEDS];

int fade_time =5; //seconds to fade between times

NeoGamma<NeoGammaTableMethod> colorGamma; // for any fade animations, best to correct gamma

//NeoPixelBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> strip(PixelCount, PixelPin);
NeoPixelBus<NeoGrbFeature, NeoWs2813Method> strip(PixelCount, PixelPin);

NeoPixelAnimator animations(AnimCount); // NeoPixel animation management object

void LoopAnimUpdate(const AnimationParam& param)
{
    // wait for this animation to complete,
    // we are using it as a timer of sorts
    if (param.state == AnimationState_Completed)
    {

    }

    float progress = NeoEase::ExponentialInOut(param.progress);
   //dump array
   for (uint16_t pixel = 0; pixel < NUM_LEDS; pixel += 1)
    { 
        float hue = 0.5f;
        float lightness;
        RgbColor color;
        
        if( panel_actual[ pixel ] == 1 ) {
          lightness = progress * MaxLightness;
          color = HslColor(hue, 1.0f, lightness);
          
          //strip.SetPixelColor(pixel, color);
          
        }
          else {
            lightness = 0;// MaxLightness - (progress * MaxLightness);
            color = HslColor(hue, 1.0f, lightness); 
            
           // strip.SetPixelColor(pixel, color);
            
          }
       strip.SetPixelColor(pixel, color);
    }
}

void setup()
{
strip.Begin();
strip.Show();

delay(5000);
hora();
animations.StartAnimation(0, 5000, LoopAnimUpdate); 
}

void loop()
{
    animations.UpdateAnimations();
    strip.Show();
}

void  hora(){

char i=180;
  RgbColor c = RgbColor(0, i, i);

panel_actual[ 101 ] = 1;
panel_actual[ 103] = 1;
panel_actual[ 102] = 1;

panel_actual[ 105] = 1;
panel_actual[ 106] = 1;
panel_actual[ 107] = 1;

panel_actual[ 53] = 1;
panel_actual[ 52] = 1;
panel_actual[ 51] = 1;
panel_actual[ 50] = 1;

panel_actual[ 39] = 1;

panel_actual[ 11] = 1;
panel_actual[ 10] = 1;
panel_actual[ 9] = 1;
panel_actual[ 8] = 1;
panel_actual[ 7] = 1;
  
}
`

Makuna, 2017-10-12:
What is your power supply rated at?

Fernando-Checa, 2017-10-12:
I use one power supply for the esp8266 (3.3v) and a 5v 10A for the strip. Common ground.
But I'm using just a few pixels, so It shouldn't be a problem.

Fernando-Checa, 2017-10-12:
Here's another example.. funLoop

[https://youtu.be/fOwbZL92WnM](url)

penfold42, 2017-10-13:
What level shifter do you have ?

Fernando-Checa, 2017-10-13:
I used a diode and also one of these:

[https://cdn.sparkfun.com//assets/parts/8/5/2/2/12009-06.jpg](url)

I will try a 74HCT245 and see if it makes any difference

penfold42, 2017-10-13:
I've not had great success with mosfet or bipolar transistor level shifters.

74HCT have always worked great for me

Fernando-Checa, 2017-10-13:
I'll just got the chip... I'll post the results as soon as I try

Fernando-Checa, 2017-10-13:
Seems to work fine..
Still problems with other libraries like fastled but ok so far with neopixelbus.

Makuna, 2017-10-13:
If you are working great, go ahead summarize what you did to fix it and I will try to incorporate the information into the "getting started" section of the Wiki that I am still working on.

Other libraries don't work correctly on Esp8266 as most use bitbang method; which just won't work on the Esp8266.  In the fastLED library issues, there is a nice write-up on stability comparisons between libraries by a third party.

Fernando-Checa, 2017-10-13:
I tried many things, but the only that really worked is using a 74HCT245N as level shifter, with 470 ohms resistor to drive the leds.
I used this wiring:

[image]

Still testing but looks really promising.

Fernando-Checa, 2017-10-20:
Also used this line to declare the new strip

`NeoPixelBus<NeoGrbFeature, NeoWs2813Method> strip(PixelCount, PixelPin);`