Makuna/NeoPixelBus #187: Gradient on RGBW strip

IVData, 2017-11-03:
I have a 5m strip of 300 SK6812 LEDs (like WS2812B).
Running the below code, the strip displays a gradient from whitish to red, instead of all being the same colour. I have tested on an Arduino Uno and ESP8266, and have also tested using SetPixelColor on each pixel individually with the same result. Using the Adafruit lib, the colours are all the same as expected.

Code:
```
#include <NeoPixelBus.h>
NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod> strip(300); // Pin added for Uno

void setup() {
  strip.Begin();
  strip.ClearTo(RgbwColor(213, 255, 222, 0));
  strip.Show();
}

void loop() {
}
```

Any ideas why this might be?

Makuna, 2017-11-03:
The issue is the WS2812b, see Wiki on [NeoWs2813Method](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neows2813method)

The newer Ws2812b use the Ws1213 timing rather than the Ws28812 timing.

Makuna, 2017-11-04:
BTW, you don't have to omit the pin for the Esp8266, it gets ignored, so if you want to sketch that works across platforms you don't have to change the code or put #ifdefs.

Makuna, 2017-12-06:
Did you get this working?

IVData, 2017-12-06:
I've had a hard time supplying higher guage wire after talking to you on gitter. Some is on the way now.
But happy that this is a voltage drop issue and not a bug so I'll close.

closed 2017-12-06 by IVData