Makuna/NeoPixelBus #191: Virtual led strips and animations

markg85, 2018-01-01:
Hi,

First of all, a big thank you to the author of this library! It really works nicely in my case. Easy to use and learn, nice C++ code (also in the backend). Thank you for making this :)

On to the feature request.
By virtual i mean "cut a strip in pieces in software and thread each piece as a individual led strip".

I am making a led light with basically 156 WRGB LED's. I've folded it in half and glued the back sides to each other. I now have a "double sided" led strip with 78 LED's on each side, but the data pin still managed all 156.

You could argue that i should use 2 PWM pins or something alike. But that would make programming it more complicated (doable, but just more tedious). So i prefer to stay away from that.

A use case i have for these virtual strips is to thread both of my LED sides as "single strips". For instance, if they are single strips in code i can just easily make a rotating circle on both sides. Right now that is.. Really tricky! I cannot use a nice feature like the rotate function as that (virtually!) rotates my whole strip (all 156 LED's, not 78!. I can set my led strip at 78 LED's which would make the first 78 work just fine, but the second 78 become more tricky to access because i then have to start making use of offsets, which the library doesn't support at all thus it becomes rather tedious to get things working.

As for the API for this, I'm thinking of something like this:

```cpp
// Remains as is
NeoPixelBus<NeoGrbwFeature, Neo800KbpsMethod> strip(158, 11);

// Arg 1: NeoPixelBus object
// Arg 2: offset
// Arg 3: Number of pixels
VirtualNeoPixelBus virtualStripOne(strip, 0, 78);
VirtualNeoPixelBus virtualStripTwo(strip, 78, 78);
```

Next there is an issue with the animator objects. A simple fix would be to allow a offset argument there as well and keep the count as is. This still leaves the freedom to execute actions on pixels in either the virtual strips or the real strip.

What do you think of this?

Best regards,
Mark

Makuna, 2018-01-06:
Did you stumble upon the NeoRingTopology?

There is a [sample](https://github.com/Makuna/NeoPixelBus/tree/master/examples/topologies/NeoPixelRingTopologyTest).  But the [RingTopology Wiki](https://github.com/Makuna/NeoPixelBus/wiki/Concentric-Rings-Support) could use some serious help.

You can provide your own layout, and consider the rings like your virtual strips.

markg85, 2018-01-06:
Ohh, that looks (from skimming over it) like the exact thing i was looking for!
Thank you very much for mentioning it. I'm gonna play with that and see if it really fits my needs :)

I will report back in here within a few days and close this issue if it does fulfill my needs.

markg85, 2018-01-14:
A "few" days later... Sorry for that :)
Yes, this indeed works as i want it to! Thank you for letting me know.

closed 2018-01-14 by markg85