Makuna/NeoPixelBus #192: Compilation error on example sketch NeoPixelCylon

pacobyte, 2018-01-05:
**Arduino IDE version:** v1.8.5
**Board:** Arduino UNO
**LED Chipset:** SK6812 RGBW, 60 count

Having just found this library today, I've been loading the examples one by one to learn how to use it. Things have been going smoothly until I tried to upload the NeoPixelCylon sketch to my Uno. What seems to be the important part of the error message is:
```
NeoPixelCylon:43: error: no match for 'operator=' (operand types are 'RgbColor' and 'Neo4Elements::ColorObject {aka RgbwColor}')
```
If you think it will help, I can include the whole compilation error output here.

Unfortunately, I've never really messed around with C/C++, so I'm clueless on how to fix this.

Makuna, 2018-01-06:
Did you compile the sketch without modification?  What modification did you make?

pacobyte, 2018-01-08:
Excellent, thank you. Changing the type of `color` to an `RgbwColor` worked perfectly.

closed 2018-01-08 by pacobyte