Makuna/NeoPixelBus #195: NeoRingTopology add "PixelInRing".

markg85, 2018-01-14:
Hi,

I have a long led strip folder over itself and divided that in 2 "virtual" strips using NeoRingTopology.
The issue i had is when making a rotating dot i need to remember the dot position.
This works for the first dot as that is the relative pixel number within the virtual strip. But you end in trouble if you increment it when it reaches the end of the virtual strip. You can't just do "pixel + 1" then as that would be outside the virtual strip. You also can't do NeoRingTopology::Map because that returns the absolute pixel.

A possible solution (and i had that working) was to check for boundaries and reset to the first pixel. It works, but makes the code longer and ugly imho. So i added a "PixelInRing" method to NeoRingTopology which is basically doing what i just said only within that class in it's own function. It's basically the same as the Map function with the difference that it returns the virtual pixel number, not the absolute one (like Map does).

The code:
```cpp
    uint16_t PixelInRing(uint8_t ring, uint16_t pixel) const
    {
        if (pixel >= getPixelCountAtRing(ring))
        {
            return 0; // invalid ring and/or pixel argument, always return a valid value, the first one
        }

        return pixel;
    }
```

Having this in the library would be helpful imho :)
I have my doubts about the name.. PixelInRing? VirtualMap? ...

Best regards,
Mark Gaiser

markg85, 2018-01-14:
Well, i have it slightly different now. Perhaps you'd like this?
I added a private method called "_inBetween" like this:
```cpp
    uint16_t _inBetween(uint16_t min, uint16_t max, uint16_t pixel, uint16_t def)
    {
      if (pixel >= min && pixel < max)
      {
        return pixel;
      }

      return def;
    }
```

And then implemented "PixelInRing" like so:
```cpp
    uint16_t PixelInRing(uint8_t ring, uint16_t pixel) const
    {
        return _inBetween(0, getPixelCountAtRing(ring), pixel, 0);
    }
```
It seems cleaner to me.
You could re-implement Map and MapProbe with this as well.

Makuna, 2018-01-18:
If you want, you can provide a pull request with the changes. 

The name might be improved but its not horrible.  An alternant could be RestrictPixelToRing().

In your case though, a better approach is to take the signed "ring relative value", increment or decrement it; then mod it with the ring size.
```
    int32 ringPixel = pixel;
    ringPixel--; // just demonstrating left rotation support and why sign is interesting here
    pixel = ringPixel % map.getPixelCountAtRing(ring);
```
This way, you could now move by two, or even three, and it willl wrap nicely.

Now with this said, your implementation is a Clamping action, while the above is a Wrapping action.  Both might be interesting.

Deep thought:  If you want an animation of the dot traveling along, then the amount of change (increment or decrement) is based on time not always a single pixel.  Thus if for some reason you could not update the animtion in time it takes it to move one pixel, it will at least flow across the strip consitently and not seem to slow.

Makuna, 2018-08-16:
https://github.com/Makuna/NeoPixelBus/pull/220

closed 2018-08-16 by Makuna