Makuna/NeoPixelBus #197: Random color flickering when setting whole strip

GregorHse, 2018-02-10:
Hello,
im using a NodeMCU ESP8266, an LedStrip with SK6812 Leds, the Arduino IDE 1.8.3 and esp2688 2.3.0 and 2.4.0(some problem with both versions).
When i run the code below, the whole strip flickers and changes between red(color it should have) and blue/purpel colors.

I've tried it with the 
Neo800KbpsMethod
NeoEsp8266Uart800KbpsMethod
NeoEsp8266Dma800KbpsMethod
NeoEsp8266Dma400KbpsMethod
NeoEsp8266Uart800KbpsMethod
Methods.

Same problem with all of them. 
I've used two Node MCUs, so i dont think my MC is damaged.

```#include <NeoPixelBus.h>

const uint16_t PixelCount = 267; 
const uint8_t PixelPin = 3;  

NeoPixelBus<NeoGrbwFeature, NeoEsp8266DmaWs2813Method> strip(PixelCount, PixelPin);

RgbwColor col(255, 0, 0, 0);

void setup()
{
    strip.Begin();
    strip.Show();
    delay(10);
}

void loop()
{
    for(int i = 0;i<PixelCount ;i++){
      strip.SetPixelColor(i, col);
     }
     delay(10);
    strip.Show();
    delay(50);
}
```

closed 2018-02-10 by GregorHse

Makuna, 2018-02-10:
Did you solve your problem?  What did you find?  I am sure others searching for solutions would like to see your discovering, even if it was something simple (as often it is).

GregorHse, 2018-02-11:
As you said, it was something simple. I had a loose contact on my data line. :)

thpallround, 2019-08-04:
> As you said, it was something simple. I had a loose contact on my data line. :)

Thank's for this issue and this answer! Hopefully, this is also my problem, back to the solder station ^^