Makuna/NeoPixelBus #199: Request: enable setting animation progress for seamless animation duration updates

nigeil, 2018-02-21:
I've been working on a program to fade a set of lights and would like to change the fade speed by pressing a button. However, the only way to do that presently is to stop the current animation and start a new one with a new duration; this restart procedure starts the animation over from the beginning, which is disruptive.

Currently, as I see it, there is no way to either:
a.) change the duration of any given animation (to speed up/slow down said animation without the need for a change in time-step), or
b.) manually set the progress of an animation after stopping the animation and starting again with a new duration passed as a parameter (to enable a 'seamless' duration change).

If there's a way to do this seamless duration change that I'm not seeing, let me know. And if not (and this is a desirable feature), I am more than willing to dig deeper and make the modifications myself before launching a pull request. 

Thanks for the awesome library!

Makuna, 2018-02-21:
You are correct, this is missing.

nigeil, 2018-02-23:
Pull request with the new functionality added and tested!
https://github.com/Makuna/NeoPixelBus/pull/201

Makuna, 2018-07-02:
I glanced at your code.  I am unsure why you exposed the setProgress and getProgress.  You don't use them in the sample and I am unsure if they would really ever be needed.  Changing the progress would cause "jumps" and thus not be seemless or smooth.

Your goal/need is to have a animation seemlessly end, thus progress to the end of the animation in a smooth but faster way than just abrouptly terminate it.  This is accomplished by just changing the duration like you have done.

Makuna, 2018-08-15:
https://github.com/Makuna/NeoPixelBus/pull/219

closed 2018-08-16 by Makuna