Makuna/NeoPixelBus #200: Error in constness of pixelIndex (NeoBuffer.h)

naelolaiz, 2018-02-21:
When trying to use NeoBuffer::GetPixelColor(x,y) I have this error:
```
In file included from .../NeoPixelBus_by_Makuna/src/NeoPixelBus.h:58:0,
                 from ....:
.../NeoPixelBus_by_Makuna/src/internal/NeoBuffer.h: In instantiation of 'typename T_BUFFER_METHOD::ColorObject NeoBuffer<T_BUFFER_METHOD>::GetPixelColor(int16_t, int16_t) const [with T_BUFFER_METHOD = NeoBufferMethod<NeoGrbFeature>; typename T_BUFFER_METHOD::ColorObject = RgbColor; int16_t = short int]':
/tmp/arduino_build_508985/sketch/letters.h:34:47:   required from here
.../NeoPixelBus_by_Makuna/src/internal/NeoBuffer.h:70:48: error: passing 'const NeoBuffer<NeoBufferMethod<NeoGrbFeature> >' as 'this' argument discards qualifiers [-fpermissive]
         return _method.GetPixelColor(pixelIndex(x, y));
                                                ^
```
That is because pixelIndex(x,y) is not const. So I fixed it with this patch:

`diff --git a/src/internal/NeoBuffer.h b/src/internal/NeoBuffer.h
index 86176c9..258285d 100644
--- a/src/internal/NeoBuffer.h
+++ b/src/internal/NeoBuffer.h
@@ -138,7 +138,7 @@ private:
 
     uint16_t pixelIndex(
         int16_t x,
-        int16_t y)
+        int16_t y) const 
     {
         uint16_t result = PixelIndex_OutOfBounds;
 
@@ -151,4 +151,4 @@ private:
         }
         return result;
     }
-};
\ No newline at end of file
+};
`

naelolaiz, 2018-02-21:
Here is the patch (I don't know how to format the text above)
[patch.txt](https://github.com/Makuna/NeoPixelBus/files/1745491/patch.txt)

Makuna, 2018-02-21:
Go ahead an make a "pull request".

Makuna, 2018-08-16:
applied changes

closed 2018-08-16 by Makuna