Makuna/NeoPixelBus #206: ESP8266 DMA method hardware WDT reset

Adam5Wu, 2018-03-31:
Thanks for this great library, very fun to work with! :)

I am creating a user configurable strip controller, and ran into a weird problem.

Background:
- Need to let user configure how many pixels
- When user change pixel count, need to blink to reflect the change
- The current implementation seems do not allow alter pixel count after instantiation
- So I work around by destroying and re-creating NeoPixelBus instances

Problem:
- When I use DMA method:
  1. I use a yield loop wait for CanShow() == true, as an indicator all DMA on old NeoPixelBus instance has completed
  2. Then I delete that NeoPixelBus instances
  3. Then I create a new NeoPixelBus instances
      - There is a high probability (~80%) of hang+hardware WDT reset in the constructor
      - I even inserted several ms delay between step 2 and 3, still same problem
- I used serial print to narrow down the point of hang, and found it was when i2c is being reset:
https://github.com/Makuna/NeoPixelBus/blob/7ebfcb7fe276c3b7409a2742093d4f479a134179/src/internal/NeoEsp8266DmaMethod.h#L228

Any idea what went wrong?

(Meanwhile I will just implement in-place alter of pixel count to bypass this problem...)

Makuna, 2018-03-31:
Could you put together a minimum sketch that demonstrates the problem (even if all it does is randomly pick different pixel counts over random amounts of time).

Also, please include what version of the Esp8266 Arduino Board support you are using.

Makuna, 2018-04-11:
I tried your sketch, I let it run for 5 minutes without an issue. 

Have you tried it on more than one Esp8266 just in case its a hardware issue?

When dealing with neopixels on projects, I have learned that in general a lot of problems arise due to power supply issues as the strip of pixels can draw a lot of amperage and this can cause problems for the esp8266 if it can't get enough power.  100 neopixels can draw 6A, and the Esp8266 when using WiFi should be provided with 0.8A.  Further complicating the power problem is that flashing the Esp8266 can cause the Pixels to flash due to shared pin and draw power while flashing that could cause flashing corruption.

Adam5Wu, 2018-04-11:
Yes, I reproduced on 4 different boards, NodeMCU, Wemos, and 2 ESP-01. It is even reproduced without actual LED attached, so power seems not a problem here.

I suspect the problem is in the framework, after several tweaking around, I found a possible culprit:
* I am using the head of the esp8266 Arduino
** There is a new feature which allows to put vtable in different locations
*** When select vtable in Flash or in IRAM, I get the hang
*** When vtable is in Heap, the sketch runs fine.

The strange thing is that, it seems to me vtable in Heap or IRAM should have no impact to ISR in terms of invocation constraints, but they did make a difference here.

My guess is that, vtable location is still not the root cause, just side effect.

Adam5Wu, 2018-04-11:
When the hang happens, esp is stuck in here:
https://github.com/Makuna/NeoPixelBus/blob/7ebfcb7fe276c3b7409a2742093d4f479a134179/src/internal/NeoEsp8266DmaMethod.h#L251

So, somehow either the DMA engine did not start properly, or the ISR did not work properly...

Adam5Wu, 2018-04-11:
It is looping the yield, expecting the dmaState to change, but it never happen.

I did additional tracing, and confirmed that when hang happens, the DMA failed to start.
1. I added a class inst counter variable, which is reset to 0 on new, and every time i2s_slc_isr() invokes increment by one;
2. I added a class static counter variable, which is initialized to 0, and every time i2s_slc_isr() invokes increment by one
3. I added Serial.print of these two counters at the beginning of the Update() and inside the yield loop.

I found, when working normally, both counters increases rapidly.
But when hang happens, the class inst counter remains 0 and class static counter remains constant.

So it means that somehow the Initialize function did not start the DMA.

Makuna, 2018-04-11:
Try adding the `volatile` qualifier to the following lines in NeoEsp8266DmaMethod.h to make them look like these
```
(line 270 ) static volatile NeoEsp8266DmaMethodBase* s_this; // for the ISR
(line 371) volatile NeoEsp8266DmaMethodBase<T_SPEED>* NeoEsp8266DmaMethodBase<T_SPEED>::s_this;
```

Adam5Wu, 2018-04-11:
`volatile` didn't help.
I also tried to check whether malloc provides aligned addresses. It did.

Makuna, 2018-04-11:
Ok, another thing to try.
Modifiy the destructor to look like this...
```
    ~NeoEsp8266DmaMethodBase()
    {
		// wait for not actively sending data
		while (_dmaState != NeoDmaState_Idle)
		{
			yield();
		}
        StopDma();
		s_this = nullptr;

        free(_pixels);
        free(_i2sBuffer);
        free(_i2sBufDesc);
    }
```
And add a check for the s_this in the ISR like this
```
if (s_this && (slc_intr_status & SLCIRXEOF))
```

Adam5Wu, 2018-04-12:
Still did not help...

When it works, I can see that some time after strip->begin(), the counters begin to increase:
```
Creating new strip! (85)
Start new strip!
Started new strip!
2 0 33969  <-- 2 = NeoDmaState_Sending
2 0 33969
2 0 33969
2 0 33969
0 152 34121 <-- 0 = NeoDmaState_Idle
0 349 34318
0 545 34514
0 742 34711
...
```

When it starts to hang, the counter never start to increase after strip->begin():
```
Start new strip!
Started new strip!
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
XD 2 0 56218
...
```

Makuna, 2018-04-12:
To be clear, you have both the changes I suggested above applied?

Also, where are these debug outputs located and what are they displaying, as you have not provided the code that includes those.

BTW, anything that is accessed in the ISR and outside the ISR MUST be marked `volatile` or they may not be the same value at both locations (due to caching and optimizations, volatile informs the compiler it can't do those things with the variable).  So those counters you added, must be marked as such.

Adam5Wu, 2018-04-12:
Yes, all counters are marked volatile, and I have integrated all changes you proposed.

As I described in earlier post, the debug output are at the beginning of update method, and inside the yield loop of update method.
The first number is the dmaState value, 0=idle, 1= pending, 2=sending. So if you see a line starts with 0, then the yield loop is never entered in that invocation of update.
The second number is per strip instance dma isr counter, it is zeroed at each new instance, and incremented once per isr invocation. This number indicates whether the isr see the correct s_this.
The third number is class static dma isr counter. It is zeroed at start of sketch, and incremented once per isr invocation. This number indicates whether isr is running at all.

Adam5Wu, 2018-04-12:
Meanwhile, I have implemented in-place pixel count change as a workaround, and it seems to work reliably. 
https://github.com/Adam5Wu/NeoPixelBus/blob/f4a28dbc2c9c18acfe0210bc62d7201a43f41ac5/src/internal/NeoEsp8266DmaMethod.h#L141-L280

I basically bypassed toggling pin mode, and reset i2s.
So whatever the problem, it seems to be in one of these operations.

Makuna, 2018-09-12:
@brendanmatkin Could you provide your config (Esp8266 Arduino board support version, hardware, etc), and try his minimal sketch above to see if it reproduces for you?  Also, are you using VTABLE is heap, IRAM, or flash?

brendanmatkin, 2018-09-12:
Thanks for the quick reply - I've been doing some testing and I think it was actually a power supply issue that was manifesting in this way. My board is custom and doesn't have much bulk capacitance because it is battery operated, but I was trying to program without the battery. 

It seemed more stable with the vtable back in heap but now that I have the battery hooked up I can't repeat the issue in heap or flash. 

Maybe that was actually @Adam5Wu 's problem too?

brendanmatkin, 2018-09-13:
> That could be possible.
> Maybe some of these pin operations cause high current spikes and
> momentrarily undervoltage condition and thus the malfunctions.
>
> But I don't have a board with high capacitance right now.
> I will later try solder a custom powersupply with big tantalum capacitor
> and try again.

Even just a 220-1000uF electrolytic across the 3v3 pins should be enough to
test the idea.

Makuna, 2018-09-20:
@Adam5Wu Another thing to try, just disconnect the pixels completely (if you still have some debugging in place to know when this happens).  This will at least remove the load they have on the power supply.

uryupinsk, 2018-11-28:
Any news on this issue? My sketch causes WDT resets randomly and very quickly and is somewhat unusable on the ESP8266, but all is fine on the ESP32 with the exact same sketch. 

I use NeoPixelBrightnessBus with DMA only to fill the LED array. FastLED is used for the animations.

Makuna, 2018-11-29:
@uryupinsk Could you provide your config? Have you tried the sketch above to see if causes a problem also?  Are you using the setting to move vtable into flash?
I am still trying to reproduce the problem.  I have not yet be able to.  As you can see above, some have confirmed it to be a power supply issue (need more current, either by cap or hefty power supply); but I have not yet discounted there is another issue; but I need to get a reproducible sketch first.

Makuna, 2019-02-05:
Some i2s cleanup has been made.

closed 2019-02-05 by Makuna