Makuna/NeoPixelBus #214: 'taskENTER_CRITICAL(mux)' is deprecated

JoanMCD, 2018-08-01:
HI, with an update of the ESP-IDF I got the following WARNING:
sketchbook\libraries\NeoPixelBus_by_Makuna\src/NeoPixelBus.h:75:0: ....
... libraries\NeoPixelBus_by_Makuna\src/internal/NeoEspBitBangMethod.h:124:13: warning: 
'taskENTER_CRITICAL(mux)' is deprecated in ESP-IDF, consider using 'portENTER_CRITICAL(mux)'

regards

Makuna, 2018-08-11:
Could you provide more details.
I suspect Esp32 although you didn't mention it.  Please confirm.
What IDE are you using, Arduino IDE or other?  What version?
What version of Esp32 Arduino board support are you using?

NOTE:  The bitbang method is not recommended for general use; as it can not provide stable data clocking.  If you are not specifically using a bit bang method like `NeoEsp32BitBang800KbpsMethod`, then you should upgrade your NeoPixelBus library as it once was the default but is no longer.

Makuna, 2018-08-12:
https://github.com/Makuna/NeoPixelBus/pull/218

closed 2018-08-16 by Makuna