Makuna/NeoPixelBus #217: Only firt 4 light tunr up WS2812B

Gedioa, 2018-08-11:
Hello all,

I have a ESP8266 , ESP-01S RGB LED, WS2812B led strip and a S - 30 - 5 30W 5V / 6A Switch Power Supply Driver. I have uploaded a script to the ESP8266 with arduino IDE. Now only the first led turn u. So i Googled em found some post that say i need a logic level converter. So i bod a LOGIC LEVEL CONVERTER BI-DIRECTIONAL from sparkfun.  I have connected the LV to the 3.3V of my raspberrypi and the HV to 5v on the Pi.  Connected the yellow wire from ESP-01S RGB LED to the LV channel of the converter and made a connection from the HV channel to the signal wire of the led strip. But even with the converter connected only 5 light turn on. I have really no idea anymore on what the problem good be.

Anyone has a solution or a idea what good be wrong in my setup.

Gedioa, 2018-08-11:
Thanks for the replyA
Yes i have looked at the wiki. (still reading some)
arduino IDE 1.8.5
ESP-01 ESP8266
I used this sketch

```
// NeoPixelFunRandomChange
// This example will randomly select a number pixels and then
// start an animation to blend them from their current color to
// randomly selected a color
// 

#include <NeoPixelBus.h>
#include <NeoPixelAnimator.h>

const uint16_t PixelCount = 300; // make sure to set this to the number of pixels in your strip
const uint8_t PixelPin = 2;  // make sure to set this to the correct pin, ignored for Esp8266

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);
// For Esp8266, the Pin is omitted and it uses GPIO3 due to DMA hardware use.  
// There are other Esp8266 alternative methods that provide more pin options, but also have
// other side effects.
//NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount);
//
// NeoEsp8266Uart800KbpsMethod uses GPI02 instead

NeoPixelAnimator animations(PixelCount); // NeoPixel animation management object

// what is stored for state is specific to the need, in this case, the colors.
// Basically what ever you need inside the animation update function
struct MyAnimationState
{
    RgbColor StartingColor;
    RgbColor EndingColor;
};

// one entry per pixel to match the animation timing manager
MyAnimationState animationState[PixelCount];

void SetRandomSeed()
{
    uint32_t seed;

    // random works best with a seed that can use 31 bits
    // analogRead on a unconnected pin tends toward less than four bits
    seed = analogRead(0);
    delay(1);

    for (int shifts = 3; shifts < 31; shifts += 3)
    {
        seed ^= analogRead(0) << shifts;
        delay(1);
    }

    // Serial.println(seed);
    randomSeed(seed);
}

// simple blend function
void BlendAnimUpdate(const AnimationParam& param)
{
    // this gets called for each animation on every time step
    // progress will start at 0.0 and end at 1.0
    // we use the blend function on the RgbColor to mix
    // color based on the progress given to us in the animation
    RgbColor updatedColor = RgbColor::LinearBlend(
        animationState[param.index].StartingColor,
        animationState[param.index].EndingColor,
        param.progress);
    // apply the color to the strip
    strip.SetPixelColor(param.index, updatedColor);
}

void PickRandom(float luminance)
{
    // pick random count of pixels to animate
    uint16_t count = random(PixelCount);
    while (count > 0)
    {
        // pick a random pixel
        uint16_t pixel = random(PixelCount);

        // pick random time and random color
        // we use HslColor object as it allows us to easily pick a color
        // with the same saturation and luminance 
        uint16_t time = random(100, 400);
        animationState[pixel].StartingColor = strip.GetPixelColor(pixel);
        animationState[pixel].EndingColor = HslColor(random(360) / 360.0f, 1.0f, luminance);

        animations.StartAnimation(pixel, time, BlendAnimUpdate);

        count--;
    }
}

void setup()
{
    strip.Begin();
    strip.Show();

    SetRandomSeed();
}

void loop()
{
    if (animations.IsAnimating())
    {
        // the normal loop just needs these two to run the active animations
        animations.UpdateAnimations();
        strip.Show();
    }
    else
    {
        // no animations runnning, start some 
        //
        PickRandom(0.2f); // 0.0 = black, 0.25 is normal, 0.5 is bright
    }
}
```

Makuna, 2018-08-12:
Glance at [this Wiki item](https://github.com/Makuna/NeoPixelBus/wiki/NeoPixelBus-object#neows2813method) and see if this fixes it.

Gedioa, 2018-08-12:
Tahnks for the reply.
I will read the wiki link you send.
When i look in the board manager and type esp8266 board i see 4 items 
Ethernet (esp8266) version 1.0.4
SD(esp8266) version 1.0.5
Servo(esp8266) version 1.0.2
Wire version 1.0.o

Somehow i think this is not what you are looking for

Greet

Makuna, 2018-08-12:
Your not looking in the board manager, what you listed is from the library manager.
The Board Manager is in a strange spot, under Tools / Board:<board selected> / Board Manager

You should see one listed as "esp8266 by ESP8266 Community".

Gedioa, 2018-08-12:
Thanks for the reply.
I look (now at the correct place) the version is 2.4.2
Also made a photo of the wiring
https://ibb.co/jPOSBU

Gedioa, 2018-08-13:
Have been playing around some more. And it turns out that Adafruit Neopixel and Neopixelbus by makuna examples don't work. Iff i use the FastLED examples the all work. 
I now have the sample code that works:

```
#include <FastLED.h>

#define LED_PIN     2
#define NUM_LEDS    300
#define BRIGHTNESS  64
#define LED_TYPE    WS2812
#define COLOR_ORDER GRB
CRGB leds[NUM_LEDS];

#define UPDATES_PER_SECOND 100

// This example shows several ways to set up and use 'palettes' of colors
// with FastLED.
//
// These compact palettes provide an easy way to re-colorize your
// animation on the fly, quickly, easily, and with low overhead.
//
// USING palettes is MUCH simpler in practice than in theory, so first just
// run this sketch, and watch the pretty lights as you then read through
// the code.  Although this sketch has eight (or more) different color schemes,
// the entire sketch compiles down to about 6.5K on AVR.
//
// FastLED provides a few pre-configured color palettes, and makes it
// extremely easy to make up your own color schemes with palettes.
//
// Some notes on the more abstract 'theory and practice' of
// FastLED compact palettes are at the bottom of this file.

CRGBPalette16 currentPalette;
TBlendType    currentBlending;

extern CRGBPalette16 myRedWhiteBluePalette;
extern const TProgmemPalette16 myRedWhiteBluePalette_p PROGMEM;

void setup() {
    delay( 3000 ); // power-up safety delay
    FastLED.addLeds<LED_TYPE, LED_PIN, COLOR_ORDER>(leds, NUM_LEDS).setCorrection( TypicalLEDStrip );
    FastLED.setBrightness(  BRIGHTNESS );
    
    currentPalette = RainbowColors_p;
    currentBlending = LINEARBLEND;
}

void loop()
{
    ChangePalettePeriodically();
    
    static uint8_t startIndex = 0;
    startIndex = startIndex + 1; /* motion speed */
    
    FillLEDsFromPaletteColors( startIndex);
    
    FastLED.show();
    FastLED.delay(1000 / UPDATES_PER_SECOND);
}

void FillLEDsFromPaletteColors( uint8_t colorIndex)
{
    uint8_t brightness = 255;
    
    for( int i = 0; i < NUM_LEDS; i++) {
        leds[i] = ColorFromPalette( currentPalette, colorIndex, brightness, currentBlending);
        colorIndex += 3;
    }
}

[... 90 lines cut ...]
    CRGB::Black
};

// Additionl notes on FastLED compact palettes:
//
// Normally, in computer graphics, the palette (or "color lookup table")
// has 256 entries, each containing a specific 24-bit RGB color.  You can then
// index into the color palette using a simple 8-bit (one byte) value.
// A 256-entry color palette takes up 768 bytes of RAM, which on Arduino
// is quite possibly "too many" bytes.
//
// FastLED does offer traditional 256-element palettes, for setups that
// can afford the 768-byte cost in RAM.
//
// However, FastLED also offers a compact alternative.  FastLED offers
// palettes that store 16 distinct entries, but can be accessed AS IF
// they actually have 256 entries; this is accomplished by interpolating
// between the 16 explicit entries to create fifteen intermediate palette
// entries between each pair.
//
// So for example, if you set the first two explicit entries of a compact 
// palette to Green (0,255,0) and Blue (0,0,255), and then retrieved 
// the first sixteen entries from the virtual palette (of 256), you'd get
// Green, followed by a smooth gradient from green-to-blue, and then Blue.
```

Makuna, 2018-08-13:
Since you state it working with FastLed, and looking at the sample code you provided, I do see an issue. 

What pin are you connecting the NeoPixels to for each sample?

The sample for FastLed you use Pin2. 

The sample for NeoPixelBus you are using GPIO3 (RDX); these are not the same pin.  

You did note that Esp8266 for my library ignores the pin you provide and always uses GPIO3 as it uses hardware to do the work which has a pin limitation?  The sample you posted above even has the comments in it stating this.

Gedioa, 2018-08-13:
Thanks for the reply.
 I did read that in ignores the PIN, but i misinterpreted it. Is there any easy way the change the pin to pin 2 in you library? Seeing that i am new to this iff it is not a easy thing to do, then i  bather lurn some more before i try it 

Again thanks for you help, as trough you comment and link i lurned some new thinks

Makuna, 2018-08-13:
You can switch to use the bitbangmethod, see wiki for esp8266 specific methods.  BUT...

If you plan to use WiFi, you will find that hardware MUST be used otherwise it will glitch.  FastLED has this problem by default (check the issues, someone has a pull request with fixes it so it uses the same hardware I use but you will then have the same problem with pins as the hardware feature only uses one pin).

Gedioa, 2018-08-13:
Thanks for the reply.
I read the wiki and now have it working, 
At the moment i don't have wifi on the ESP so for now i can play around with it.
I am now running you funradomchange  runs like a charm.

Again thanks for helping out. 
I really appreciated it.

closed 2018-08-13 by Gedioa