Makuna/NeoPixelBus #229: Specific SK6812 support

Makuna, 2018-08-23:
https://cdn-shop.adafruit.com/product-files/1138/SK6812+LED+datasheet+.pdf

While the WS2813 will work, the reset time is longer than required.  

While the default can work in some sketches, the reset time is shorter than required.

r1dd1ck, 2018-08-24:
Your current `default` of **50µs** is simply too short. While it might work most of the time for older chips, it still is just barely "on edge" even there. Should be changed to at least **80µs+** at minimum to conform to a wider range. Or even straight to **300µs**, which would meet requirements of all chips currently available.

And I think that "longer than required" is not much of an argument here. Whether it is 80 or 300µs makes little difference to the resulting frame rate, unless 1000+ FPS is your target.

r1dd1ck, 2018-08-24:
> but where is the data backing this up?

most of them are directly available on chip manufacturer's websites. 

Just look at the domain names of your "sources" (rolleyes) .. 
I don't see a single one belonging to any genuine chip manufacturer. Those are all reseller websites, with mostly outdated datasheets. And to make matters worse - it's often just the very first revisions of those datasheets, because the resellers did not bother to update them since product release..

Which gives?
... most of the values you posted above are not in accord with latest manufacturer recommendations speciffied in latest revisions of corresponding datasheets.

Here you go: [LED datasheets (updated 2018/May)](https://file.io/Rdnajk)

> Wasting cycles is just wasting cycles

No cycles are being wasted here. Reset times are just a grace period after each frame refresh, and even in worst-case scenarios (300µs) have close to zero impact for all usual use cases (FPS < 400). However I agree that it is prefferable to not necessarily overdo it for cases where it is not required. But still - I think that `defaults` should be `SAFE`.

> I have been wanting to expand the "methods" list to include the specific model, this already started with the WS2813 and I will continue to expand it.

Thumbs up on this. Giving the end-user more options to choose from is usually a good thing :-)

Makuna, 2018-08-24:
Your link doesn't work, 404 error'; so its not useful yet.  I searched for datasheets and FOUND nothing from any manufacturer; so I can't use something I don't have.

r1dd1ck, 2018-08-24:
Arrgh.. those free file hosters :-) Was working when I put it up a couple hours ago..

Take # 2:
[LED datasheets (updated 2018/May) - file.io](https://file.io/v3mSDa)
[LED datasheets (updated 2018/May) - zippy](https://www21.zippyshare.com/v/tCM1buUt/file.html)

> I searched for datasheets and FOUND nothing from any manufacturer;

Well, I don't know where you have been looking, but was probably not the internet :-D
Try this: [World-Semi](http://www.world-semi.com/solution/list-4-1.html) :-)

Makuna, 2018-09-21:
Thanks, I will be refactoring the methods a little on the next update. I hate that they reused the same model number but changed the specs.  That's bad form.   What makes it worse, is that I have plenty of Ws2812b that work fine with just the 50us delay even though some specs show 300us (and the ones you pointed me to state >280us).  And I do know others that have Ws2812b that do not work with just 50us.

r1dd1ck, 2018-09-21:
Yes, the later revisions require longer reset times, so they just decided to settle the recommended value @ **>280µs** for all of them. Maybe a special flag to enable "ultra-short" reset times could be introduced.

Re-factoring & unification of the speed methods into a single global/universal model is the next best thing to do. Although modifications to most of the output methods will be required - it's entirely worth it in the long run.

Makuna, 2018-09-21:
Actually, its not just later revisions.  It turns out two companies are making them, both used the same model names, each had different timing requirements.  One company is shifting away to different model names now.  p.s.  FLAGS ARE BAD (as far as this library and my coding style), but maybe you meant that in a general sense like default method using the slower reset and another one using a faster reset; which is the current plan.

Makuna, 2018-09-22:
https://github.com/Makuna/NeoPixelBus/pull/230

closed 2019-02-05 by Makuna