Makuna/NeoPixelBus #236: Quck updates

nenadalm, 2018-11-19:
**Describe the bug**
When I try to change pixels quickly with new instance of the bus, 2nd change is not taken into account.

**To Reproduce**
```c++
#include <NeoPixelBus.h>

#define STRIP_PIN 22

NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod>* strip = NULL;

void replace_strip(uint16_t count) {
  if (strip) {
    if (strip->PixelCount() == count) {
      return;
    }

    delete strip;
  }

  strip = new NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod>(count, STRIP_PIN);
  strip->Begin();
}

void first_pattern() {
  replace_strip(30);
  RgbColor color0(10, 10, 10);
  strip->SetPixelColor(0, color0);
  RgbColor color14(10, 10, 10);
  strip->SetPixelColor(14, color14);
  RgbColor color29(10, 10, 10);
  strip->SetPixelColor(29, color29);
  strip->Show();
}

void second_pattern() {
  replace_strip(3); // if new instance is not created (this is commented out), it works
  RgbColor color0(0, 10, 0);
  strip->SetPixelColor(0, color0);
  RgbColor color1(0, 10, 0);
  strip->SetPixelColor(1, color1);
  RgbColor color2(0, 10, 0);
  strip->SetPixelColor(2, color2);
  strip->Show();
}

void program_test() {
  // init
  replace_strip(30);
  strip->Show();

  first_pattern();
// it works with delay
//  delay(1000);
  second_pattern();
}

void setup() {
  program_test();
}
```

**Expected behavior**
Leds from `second_pattern` should be on in specified colors.

**Development environment (please complete the following information):**
 - OS: Fedora 28
 - Build Environment Arduino IDE v1.8.7
 - Board target ESP 32 Dev Module
 - Library version v2.3.4

Makuna, 2018-11-19:
Changed the title to be more specific.
Could you provide the exact pixels and approximate colors you are seeing.

nenadalm, 2018-11-19:
I see what would be shown if `second_pattern()` wasn't there at all.

nenadalm, 2018-11-19:
So I see 3 leds (0, 14, 29) that has dim white (10, 10, 10) color. (I will recheck tomorrow, but I think that's how it was).

Makuna, 2018-11-20:
I am remote with hardware, so I can't test this for a week.  But here are a few things ...

First, a few things for diagnosing; 
* use a color that is a little brighter, use 30 instead of 10.   
* note, that when the strip is switched to less pixels, it will not reset those pixels beyond its count.  So pixel 14 and 29 should remain on until you show a strip with a count that includes those.  SO, put a delay in the loop to see what happens between loops
```
void setup() {
  program_test();
delay(5000); // to see if the issue is just within the program_test quick updates
}
```
Since you are dynamically allocating; add the following into your replace_strip function to make sure the strips data is fully sent before they get deleted (since the data is sent async by hardware) 
```
  if (strip) {
    if (strip->PixelCount() == count) {
      return;
    }
    while (!strip->CanShow()) { yield(); }; // add this line
    delete strip;
  }
```

nenadalm, 2018-11-20:
I am remote from the hardware too currently (not sure when it be available again to me).

I've tried blocking
```c++
while (!strip->CanShow());
```
as you mentioned and it didn't help.

For now I solved it by allocating maximum supported number of pixels at the beginning without changing it.

yuranikspb, 2020-10-01:
i have same question how,create dynamic instancing of NeoPixelBus template without argumets
i want to change led type dynamic,

Makuna, 2020-10-01:
@yuranikspb Please don't jump on existing seemly related issue to ask questions.  Gitter is where you ask questions.
The short answer is you can't.  This is how templates works; they are different types.

Makuna, 2021-08-09:
I just merged some cleanup with this pull #511; get the latest and give it a try.

closed 2021-10-12 by Makuna