Makuna/NeoPixelBus #240: Cannot compile with 2.4.0

empty88, 2018-12-24:
Hey there,

since Version 2.4.0 I cannot compile anymore, this is a sample project to reproduce this issue, with version 2.3.5 there is no problem:
`#include <NeoPixelAnimator.h>
#include <NeoPixelBrightnessBus.h>

#define pixelCount 114            //number of pixels in RGB strip

NeoPixelBrightnessBus<NeoGrbwFeature, NeoSk6812Method> strip(pixelCount, 0);  //GPIO 0

const uint8_t PanelWidth = 11;  // 8 pixel x 8 pixel matrix of leds
const uint8_t PanelHeight = 10;
const uint8_t TileWidth = 1;  // laid out in 4 panels x 2 panels mosaic
const uint8_t TileHeight = 1;

typedef ColumnMajorAlternating270Layout MyPanelLayout;
typedef ColumnMajorLayout MyTilesLayout;
NeoTiles <MyPanelLayout, MyTilesLayout> tiles(PanelWidth,PanelHeight,TileWidth,TileHeight);

void setup() {
  // put your setup code here, to run once:

  strip.Begin();
  strip.Show();
}

void loop() {
  // put your main code here, to run repeatedly:

}`

Error:
C:\Users\Marco\Nextcloud\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.cpp: In static member function 'static void NeoEsp8266UartInterruptContext::Isr(void*)':

C:\Users\Marco\Nextcloud\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.cpp:112:22: error: static void NeoEsp8266UartInterruptContext::Isr(void*) causes a section type conflict with static const volatile uint8_t* NeoEsp8266UartContext::FillUartFifo(uint8_t, const volatile uint8_t*, const volatile uint8_t*)

 void ICACHE_RAM_ATTR NeoEsp8266UartInterruptContext::Isr(void* param)

                      ^

In file included from C:\Users\Marco\Nextcloud\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.cpp:28:0:

C:\Users\Marco\Nextcloud\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.h:50:52: note: 'static const volatile uint8_t* NeoEsp8266UartContext::FillUartFifo(uint8_t, const volatile uint8_t*, const volatile uint8_t*)' was declared here

     static const volatile uint8_t* ICACHE_RAM_ATTR FillUartFifo(uint8_t uartNum, 

                                                    ^

Makuna, 2018-12-24:
Interesting, 2.5.0b2 also compiles fine.  Something with 2.4.0 is causing an issue.

Makuna, 2018-12-25:
https://github.com/Makuna/NeoPixelBus/pull/241

Makuna, 2018-12-25:
new build 2.4.1 fixes the issue with using esp8266 Arduino board support 2.4.2 release.

closed 2018-12-25 by Makuna