Makuna/NeoPixelBus #243: NeoEsp8266UartMethod.h error swap is not a member of std

Wifiunterputzdose, 2019-01-04:
Hello, I'm using the ESP8266HUEEMulator with the NeoPixelBusLib Version 2.4.1

After compiling in PlatformIO I got this compile error

`In file included from C:\Users\xxx\.platformio\lib\NeoPixelBus_ID547\src\internal\NeoEsp8266UartMethod.cpp:28:0:
C:\Users\xxx\.platformio\lib\NeoPixelBus_ID547\src\internal\NeoEsp8266UartMethod.h: In member function 'void NeoEsp8266AsyncUart
<T_UARTFEATURE, T_UARTCONTEXT>::UpdateUart()':
C:\Users\xxx\.platformio\lib\NeoPixelBus_ID547\src\internal\NeoEsp8266UartMethod.h:260:9: error: 'swap' is not a member of 'std'
std::swap(_asyncPixels, _pixels);
^
*** [.pioenvs\nodemcuv2\lib97a\NeoPixelBus_ID547\internal\NeoEsp8266UartMethod.cpp.o] Error 1
 [ERROR] Took 14.53 seconds
 `

How to get rid of this error?

Makuna, 2019-01-04:
Include code enough to give some sort of context please.

Are you using NeoEsp8266Uart1Method or NeoEsp8266Uart0Method?

T3rorx, 2019-01-05:
I have the same issue !

Makuna, 2019-01-05:
Try changing the line
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart800KbpsMethod> 
```
to 
```
NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart1Ws2812xMethod> 
```

Wifiunterputzdose, 2019-01-06:
Hi Michael thanks for your help.

with an older version 2.3.3 I get rid of this error. 

I will give a try with the new version and your suggestion

ScS123, 2019-01-07:
I have long used:
NeoPixelBus<NeoGrbFeature, NeoEsp8266Dma800KbpsMethod> strip(PixelCount, PixelPin);
for my esp8266 12E builds through NeoPixelBus 2.4.0. It has always given me solid service.

As of version 2.4.1 I also get the following compile error:
In file included from C:\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.cpp:28:0:

C:\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.h: In member function 'void NeoEsp8266AsyncUart<T_UARTFEATURE, T_UARTCONTEXT>::UpdateUart()':

C:\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.h:260:9: error: 'swap' is not a member of 'std'

         std::swap(_asyncPixels, _pixels);

Using "NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart1Ws2812xMethod> " also gives me the same error.

ScS123, 2019-01-07:
In all fairness, I looked at my other build. Running Arduino IDE 1.8.7 with Makuna NeoPixelBus 2.4.1 compiling SAME code just fine. Installed depricated Arduino IDE 1.8.7 on my first machine and still have same issues with NeoPixelBus 2.4.0 and above. Obviously missing some library somewhere....

Makuna, 2019-01-07:
std::swap is part of the stl files; which gets included with the Esp8266 board support. In Arduino.h on the line `#include <algorithm>` should bring it in.

What version of the esp8266 board support are you using (look in the board manager).
I am using:
Arduino IDE 1.8.5
Board Support 2.5.0.rc2
NeoPixelBus version shouldn't matter on this single line, as swap has been present in the code for several years.

I will try the latest Arduino IDE and see if I can repro with it.
Its an evironment config issue then.

Makuna, 2019-01-07:
Ok, I can't reproduce this all with Arduino IDE 1.8.8 (latest)
There is more than likely a configuration issue with your IDE evironment.
I would recommend you close the IDE, then clear your temp files (C:\\Users\\<user name>\\AppData\\Local\\Temp\\arduino_build_*) and then start the IDE.

ScS123, 2019-01-07:
After much searching, the problem stemmed from using a "staging" version of the ESP8266 board definition. After changing to the "stable" version and recompiling, there were no issues with 2.4.1

Makuna, 2019-01-11:
@Wifiunterputzdose Did you see ScS123 solution?

closed 2019-01-20 by Makuna

MBerka, 2020-01-06:
I just encountered this with a regular version of PIO (Staged must have eventually become Stable). The problem occurs because the framework's Arduino.h file ([copy](https://github.com/choey/framework-arduinoespressif8266/blob/master/cores/esp8266/Arduino.h)) does not have `#include <algorithm>`. However, I found that NeoEsp8266UartMethod.cpp explicitly includes <utility>, which [also defines std::swap](https://en.cppreference.com/w/cpp/algorithm/swap). PlatformIO compilation completes when `#include <utility>` is moved from NeoEsp8266UartMethod.cpp to NeoEsp8266UartMethod.h.

https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/Arduino.h