Makuna/NeoPixelBus #244: DotStarSpiMethod should support setting the SPI pins (ESP32)

debsahu, 2019-01-14:
On ESP32, hardware pins are selectable. Which means for **DotStarSpiMethod**, users should be able to select the pins they want to use. This requires changes to this code for exposing the pin parameters.

The default pins for hardware SPI on ESP32 are : **GPIO18(SCK)** is **CLK**, **GPIO23(MOSI)** is **DATA**
One can check this for their board:
```
Serial.print("Hardware SPI //DATA PIN: ");
Serial.println(MOSI); //GPIO23?
Serial.print("Hardware SPI //CLOCK PIN: ");
Serial.println(SCK);  //GPIO18?
```

In the meantime, users can use slower software SPI method: **DotStarMethod** where pins can be defined. 
```
//Software SPI method: Any pin can be clock and data
#define PIN_CLK 18
#define PIN_DATA 23
DotStarMethod dma = DotStarMethod(PIN_CLK, PIN_DATA, ledCount, 3);
```

Makuna, 2019-02-01:
This is pending the next release.

debsahu, 2019-02-03:
Also a good reference for future: https://learn.adafruit.com/using-atsamd21-sercom-to-add-more-spi-i2c-serial-ports/creating-a-new-spi

closed 2019-02-05 by Makuna