Makuna/NeoPixelBus #248: Stack canary watchpoint triggered when in task ESP32

seflm, 2019-01-21:
ESP32 crashing when calling strip.show() from RTOS task running on core 0. Using pin 25 with 4 WS2812B LEDs.

**Crash**
`PC: 0x400892a3
EXCVADDR: 0x00000000

Decoding stack results
0x400874a0: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/queue.c line 1562
0x400d18e9: i2sWrite at C:\Users\marek\Documents\Arduino\libraries\NeoPixelBus-master\src\internal\Esp32_i2s.c line 468
0x400d0d69: NeoEsp32I2sMethodBase ::Update() at C:\Users\marek\Documents\Arduino\libraries\NeoPixelBus-master\src/internal/NeoEsp32I2sMethod.h line 141
0x400d0d7e: NeoPixelBus   >::Show() at C:\Users\marek\Documents\Arduino\libraries\NeoPixelBus-master\src/NeoPixelBus.h line 147
0x400d0e2a: code_breath(void*) at C:\Users\marek\AppData\Local\Temp\arduino_modified_sketch_373649/sketch_jan21a.ino line 35
0x40089589: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143`

Example attached
[issue_example.zip](https://github.com/Makuna/NeoPixelBus/files/2777258/issue_example.zip)

corndog2000, 2019-11-19:
Having the same issue when using 130 leds.

Makuna, 2019-11-19:
There are no task protections for the show call.  I don't believe this has anything to do with the number of leds.
There is an assumption (its in the documents) that you always call `show` within `setup` or `loop`. You need to manage any task related semaphore/core issues.
Also note, the default method right now is RMT (channel 6 I believe) and not i2s.

Makuna, 2023-07-09:
ESP32 I2S has been rewritten since this issue was generated.  I had never duplicated this issue and I have had no other reports of this since then, so either it was fixed with the rewrite or one of several IDF updates.

closed 2023-07-09 by Makuna