Makuna/NeoPixelBus #252: Latest ESP8266 Arduino Core causing error

debsahu, 2019-02-12:
With this inclusion https://github.com/esp8266/Arduino/blob/master/cores/esp8266/esp8266_undocumented.h in latest arduino esp8266 core, the following line:
https://github.com/Makuna/NeoPixelBus/blob/9762439c85e5add538ee37c2ba6a80c146942de0/src/internal/NeoEsp8266DmaMethod.h#L51

gives this error:
```
.piolibdeps\NeoPixelBus_ID547\src/internal/NeoEsp8266DmaMethod.h:51:127: error: declaration of C function 'void rom_i2c_writeReg_Mask(uint32_t, uint32_t, uint32_t, uint32_t, uint32_t, uint32_t)' conflicts with
void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);
^
In file included from C:\Users\TTTTTTT\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/esp8266_peri.h:25:0,
from C:\Users\TTTTTTT\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/Arduino.h:38,
from c:\users\TTTTTTT\appdata\local\temp\1\tmp_wfgu5:1:
C:\Users\TTTTTTT\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266/esp8266_undocumented.h:9:13: error: previous declaration 'void rom_i2c_writeReg_Mask(int, int, int, int, int, int)' here
extern void rom_i2c_writeReg_Mask(int, int, int, int, int, int);
^
*** [.pioenvs\nodemcuv2\src\XXXXXXXXX.ino.cpp.o] Error 1
```

debsahu, 2019-02-12:
Commenting out  `//void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);` from `NeoPixelBus/src/internal/NeoEsp8266DmaMethod.h` fixes the compilation error. Maybe adding some kind of compilation flag would help as this has been introduced in last 6 days.

Makuna, 2019-02-14:
v2.5.0 doesn't use the latest core and this is not an issue, yet.  You are working on the bleeding edge when you get the latest core ;-)
As far as I know, I can't find a symbol to use to put a conditional around it; and C++ doesn't support this using a the existence of a function in the conditionals.

Makuna, 2019-02-14:
Could you try this out?

wrap that line in NeoEsp8266DmaMethod.h so this ...
```
void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);
```
becomes this...
```
#if !defined(__CORE_ESP8266_VERSION_H)
    void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);
#endif
```
If it works I will create a pull and merge it in.

debsahu, 2019-02-14:
@Makuna I will test it today and let you know. It should work.

debsahu, 2019-02-14:
yes works! thanks.

debsahu, 2019-02-14:
__CORE_ESP8266_VERSION_H is defined in core v2.5.0

Would this be a better approach:
```
#if ARDUINO_ESP8266_GIT_VER != 0x00000000
    void rom_i2c_writeReg_Mask(uint32_t block, uint32_t host_id, uint32_t reg_add, uint32_t Msb, uint32_t Lsb, uint32_t indata);
#endif
```

Makuna, 2019-02-14:
That way isn't future proof.  When the next release comes out I need this to go away.
This solution was suggested by the esp devs.

NOTE:  The latest NeoPixelBus library released version is not guaranteed to be backward compatible with previous to its release versions of ANY board support version.  Since 2.5.0 is out, my next version which will include these changes will then make this a requirement.

Makuna, 2019-02-15:
https://github.com/Makuna/NeoPixelBus/pull/254

closed 2019-02-25 by Makuna