Makuna/NeoPixelBus #256: 'NEO_GRB' was not declared in this scope

wfraga18, 2019-03-05:
Hey Makuna,

Thank you by the Lib, I'm trying to use, by the first time, and I need  use NeoPixelMatrix too, but I didn't find nothing to work with your lib, than I'm supposing that I can use the adafruit lib
but I'm getting: 'NEO_GRB' was not declared in this scope

Do you know a workaround to use Matrix Lib ?
thank you

wfraga18, 2019-03-05:
what is the best way to integrate with neopixelbus  and
<Adafruit_NeoMatrix.h>
<Adafruit_GFX.h>
thanks
*using esp8266

Makuna, 2019-03-05:
Please honor the instructions given when creating an issue.  This is not the place to ask questions; this is a place to track bugs in the code.

Refer to the Wiki and Readme for the link for the gitter channel where questions can be asked.

closed 2019-03-05 by Makuna