Makuna/NeoPixelBus #258: HtmlColor::Parse fails to compile

skyhisi, 2019-03-13:
`HtmlColor::Parse` doesn't compile.

**Example sketch**
```c++
#include <NeoPixelBus.h>

void setup()
{
  HtmlColor colour;
  colour.Parse<HtmlShortColorNames>("red");
}

void loop()
{}
```
**Compiler error**
```text
In file included from /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/stdio.h:6:0,
                 from /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/stdio.h:63,
                 from /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Arduino.h:32,
                 from /home/pi/Arduino/NeoPixelBusParseBug/build/sketch/NeoPixelBusParseBug.ino.cpp:1:
/home/pi/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/HtmlColor.h: In instantiation of 'size_t HtmlColor::Parse(const char*, size_t) [with T_HTMLCOLORNAMES = HtmlShortColorNames; size_t = unsigned int]':
/home/pi/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/HtmlColor.h:256:73:   required from 'size_t HtmlColor::Parse(const char*) [with T_HTMLCOLORNAMES = HtmlShortColorNames; size_t = unsigned int]'
/home/pi/Arduino/NeoPixelBusParseBug/NeoPixelBusParseBug.ino:6:42:   required from here
/home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/tools/sdk/libc/xtensa-lx106-elf/include/sys/pgmspace.h:76:82: error: 'const void*' is not a pointer-to-object type
     #define pgm_read_ptr(addr)              (*reinterpret_cast<const void*>(addr))
                                                                                  ^
/home/pi/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/HtmlColor.h:215:47: note: in expansion of macro 'pgm_read_ptr'
                     PGM_P searchName = (PGM_P)pgm_read_ptr(&colorPair->Name);
                                               ^
Using library NeoPixelBus_by_Makuna at version 2.4.3 in folder: /home/pi/Arduino/libraries/NeoPixelBus_by_Makuna 
Using library SPI at version 1.0 in folder: /home/pi/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/libraries/SPI
```

**Development environment:**
 - OS: Linux / Raspian / Debian Stretch
 - Build Environment: Arduino 1.8.8, ESP8266 2.5.0, using `arduino-builder`
 - Board target: esp8266:esp8266:d1
 - Library version: 2.4.3

Makuna, 2019-03-14:
This is a bug in Esp8266 2.5.0.  
Your example compiles with every other platform and even with esp8266 v 2.4.2.

Please create an issue in that github project.

Makuna, 2019-03-14:
This was already fixed and will be in the next release.

https://github.com/esp8266/Arduino/issues/5733

To work around the problem, you can apply what I listed above

closed 2019-03-14 by Makuna

skyhisi, 2019-03-14:
Thanks for looking at this, sorry I didn't think to check the older version of the board library.