Makuna/NeoPixelBus #259: 8266 Neo800KbpsMethod does not update on Show from interrupt handler

BbIKTOP, 2019-03-18:
I use nodeMCU and also tried bare esp8266-12F with the same results.
I have declader NeoPixelBus:
`int pixelsCount = 144 * 5;
NeoPixelBus<NeoGrbFeature, Neo800KbpsMethod> strip(pixelsCount, 3);`

I have interrupt handler called for rotary encoder:
`void setLedBrightness(int value)
{
  if (value > 255) value = 255;
  if (value < 0)value = 0;

  Serial.println("Color will be set to " + String(value));

  RgbColor current(value, value, value);
  strip.ClearTo(current, 0, currentPixelsCount);
  Serial.println("CanShow=" + String(strip.CanShow()) + ", total pixels=" + String(currentPixelsCount));

  if (strip.CanShow())
  {
    strip.Show();
  }
  else
  {
    Serial.println("!!!!!! CanShow returned false, won't show anything now");
  }

  analogWrite(5, value);
  Serial.println("Color has been set to " + String(value));
}

void processRotation(Encoder & e, int direction, int buttonPressed)
{
  String sDir = direction > 0 ? "+" : "-";
  String sButton = buttonPressed ? " [X]" : " [ ]";

  Serial.println("Encoder: Rotated " + sDir + sButton + ", current value=" + String(e.getValue()));

  if (buttonPressed)
  {
    currentPixelsCount += direction;
    if (currentPixelsCount < 1) currentPixelsCount = 1;
    if (currentPixelsCount > pixelsCount) currentPixelsCount = pixelsCount;
    Serial.println("Current pixels count=" + String(currentPixelsCount));
  }

  long value = e.getValue() * 256l / e.getMaxValue();
  setLedBrightness(value);
}
`
I also connected singleordinary led to pin 5 to control the process.
Now, when I see in the Serial output that interrupt is called and new value set, it also sets the led brightness as well as led brightness. The ordinary led works ok, but strip works just once, then does not update for a few seconds then start working again.

I use ESP8266version 2.5.0 with arduino ide 1.8.8 under macos 10.14.3, NeoPixelBus ver 2.4.3 downloaded via library manager

Also, when I use 
`NeoPixelBus<NeoGrbFeature, NeoEsp8266Uart1800KbpsMethod> strip(pixelsCount, 3);`
everything works fine.

[NeoPixelBusInterrupt.tar.gz](https://github.com/Makuna/NeoPixelBus/files/2979940/NeoPixelBusInterrupt.tar.gz)

Makuna, 2019-03-18:
There is nothing I can do about this.  The Show method will call lower level hardware features, which should never be called from an ISR.  It is normal on esp8266 to never call anything that accesses the hardware from an ISR other than the hardware associated with the ISR.  Further, Show will block until the last send is finished, which isn't what you will want to do either.

Best practice is to set a flag (volatile global variable) that is checked in the loop.  Something like this...

```
volatile uint8_t pixelShowFlag;

void isrCallback() {
{
pixelShowFlag = true;
}

void setup() {
pixelShowFlag = false;
}

void loop {
if (pixelShowFlag) {
    strip.Show();
}
}
```

closed 2019-03-18 by Makuna

BbIKTOP, 2019-03-18:
Got it, thank you. But uart method works fine from the isr - can I rely on it or rather use polling, either for strip or for my routines?