Makuna/NeoPixelBus #261: Compilation error for examples on ESP8266 (Wemos D1 mini)

FrankenApps, 2019-04-05:
Arduino 1.8.5 on Windows 10
Wemos D1 mini
Whenever I try to compile any of the example sketches, I get the following compiler error:
`Arduino: 1.8.5 (Windows 10), Board: "WEMOS D1 mini, 160 MHz, 80MHz, DIO, 115200, 1M (256K SPIFFS)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Me\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Me\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=deauther:esp8266:d1_mini:CpuFrequency=160,FlashFreq=80,FlashMode=dio,UploadSpeed=115200,FlashSize=1M256 -ide-version=10805 -build-path C:\Users\FERDIN~1\AppData\Local\Temp\arduino_build_837019 -warnings=none -build-cache C:\Users\FERDIN~1\AppData\Local\Temp\arduino_cache_951608 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\FERDIN~1\AppData\Local\Temp\arduino_modified_sketch_714362\NeoPixelRotateLoop.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Me\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Me\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Me\Documents\Arduino\libraries -fqbn=deauther:esp8266:d1_mini:CpuFrequency=160,FlashFreq=80,FlashMode=dio,UploadSpeed=115200,FlashSize=1M256 -ide-version=10805 -build-path C:\Users\FERDIN~1\AppData\Local\Temp\arduino_build_837019 -warnings=none -build-cache C:\Users\FERDIN~1\AppData\Local\Temp\arduino_cache_951608 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\0.1.2 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.esptool.path=C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\esptool\0.4.6 -verbose C:\Users\FERDIN~1\AppData\Local\Temp\arduino_modified_sketch_714362\NeoPixelRotateLoop.ino
Using board 'd1_mini' from platform in folder: C:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther
Using core 'esp8266' from platform in folder: C:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther
Detecting libraries used...
"C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include" -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\variants\d1_mini" "C:\Users\FERDIN~1\AppData\Local\Temp\arduino_build_837019\sketch\NeoPixelRotateLoop.ino.cpp" -o "nul"
"C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include" -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\variants\d1_mini" "-IC:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "C:\Users\FERDIN~1\AppData\Local\Temp\arduino_build_837019\sketch\NeoPixelRotateLoop.ino.cpp" -o "nul"
"C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include" -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\variants\d1_mini" "-IC:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\SPI" "C:\Users\FERDIN~1\AppData\Local\Temp\arduino_build_837019\sketch\NeoPixelRotateLoop.ino.cpp" -o "nul"
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\Esp32_i2s.c
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HsbColor.cpp
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HslColor.cpp
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HtmlColor.cpp
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HtmlColorNameStrings.cpp
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HtmlColorNames.cpp
Using cached library dependencies for file: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\HtmlColorShortNames.cpp
"C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include" -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\cores\esp8266" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\variants\d1_mini" "-IC:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src" "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\SPI" "C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna\src\internal\NeoEsp8266UartMethod.cpp" -o "nul"
"C:\Users\Me\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther/tools/sdk/include" -c -Os -g -fno-threadsafe-statics -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=160000000L -DARDUINO=10805 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266  -DESP8266 "-IC:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardwa [... cut ...]
     uint8_t avail = (UART_TX_FIFO_SIZE - GetTxFifoLength(uartNum)) / 4;

                      ^

Bibliothek NeoPixelBus_by_Makuna in Version 2.4.4 im Ordner: C:\Users\Me\Documents\Arduino\libraries\NeoPixelBus_by_Makuna  wird verwendet
Bibliothek SPI in Version 1.0 im Ordner: C:\Users\Me\AppData\Local\Arduino15\packages\deauther\hardware\esp8266\2.0.0-deauther\libraries\SPI  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board WEMOS D1 mini.`
This is the full Traceback, I'm sorry it is in german.

Makuna, 2019-04-05:
Please include ALL versions of pieces.  You didn't mention which board support version (see board manager) and which version of the NeoPixelBus library (see Library Manager).

Also, update your Arduino IDE, they often induce small bugs that effect compilation then quickly fix it with a version update.  I am running 1.8.9 and don't see this problem.

FrankenApps, 2019-04-05:
Well, I resolved my issue. It was not a problem with the library.
I had another esp8266 SPI library installed on my system, that was used instead of the one installed by the board manager.
The Arduino IDE actually told me in the third last line of the error log, I just did not see it.
Sorry for the inconvenience.

closed 2019-04-05 by Makuna