Makuna/NeoPixelBus #262: strip.Show stops working even if the rest of the sketch runs further

chekaaaaa, 2019-04-07:
I have made a small sketch which connects to the wifi and make requests to an endpoint on my server the Server returns this array:

`{"anObject":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,15277667,0,0,15277667,0,0,0,15277667,15277667,15277667,15277667,15277667,15277667,0,0,0,15277667,15277667,15277667,15277667,0,0,0,0,0,15277667,15277667,0,0,0,0,0,0,0,0,0,0,0,0,15277667,15277667,15277667,15277667,15277667,15277667,0,0,0,0,0,0,0,0,0,0,0,15277667,0,0,15277667,0,0,0,15277667,15277667,15277667,15277667,15277667,15277667,0,0,0,15277667,15277667,15277667,15277667,0,0,0,0,0,15277667,15277667,0,0,0,0,0,0,0,0,0,0,0,0,15277667,15277667,15277667,15277667,15277667,15277667,0,0,0,0,0,0,0,0,0]}`

The arrays contains two images so i loop two times over it but after 64 pixels I reset the pixel number after a delay of one second so i "render" the second half of the array. 

This works fine but only for two-three minutes. After this the sketch is still executing. It still puts the http requests and the print statement is fired before and after the  strip.Show Statement. But the leds are getting dark and dont start working until i reset the controller. Than the sketch works for 3 minutes and suddenly the leds are getting dark again....

Here ist my Code:
```
#include <ArduinoJson.h>
#include <WiFi.h>
#include <HTTPClient.h>
#include <NeoPixelBrightnessBus.h>
 
const char* ssid = "inser tyour SSID";
const char* password =  "your PW";

const int PixelCount = 64; 
const int PixelPin = 5;
int ledIndex = 0;

const size_t capacity = JSON_ARRAY_SIZE(32) + JSON_OBJECT_SIZE(1) + 290;
DynamicJsonBuffer jsonBuffer(capacity);

NeoPixelBrightnessBus<NeoGrbFeature, Neo800KbpsMethod> strip(PixelCount, PixelPin);

 void setup() {
  Serial.begin(115200);
  delay(4000);
  WiFi.begin(ssid, password);
  strip.Begin();
    while (WiFi.status() != WL_CONNECTED) {
    delay(1000);
    Serial.println("Connecting to WiFi..");
  }
  Serial.println("Connected to the WiFi network");
  
}
 
void loop() {
 
  if ((WiFi.status() == WL_CONNECTED)) { //Check the current connection status
 
    HTTPClient http;
    http.setReuse(true);
    http.begin("http://bleedingedge.onl:8080/chunked");
 
    int httpCode = http.GET();

    Serial.print(httpCode);
    if (httpCode > 0) { //Check for the returning code
 
        String payload = http.getString();
        Serial.println(payload);
        JsonObject& root = jsonBuffer.parseObject(payload);

        JsonArray& anObject = root["anObject"];
        http.end();
           for( int i = 0; i < PixelCount * 2; i++) {
            int htmlColor = anObject[i];
            strip.SetPixelColor(ledIndex, HtmlColor(htmlColor)); 
            ledIndex++;   
              
             if(ledIndex == PixelCount){
              Serial.println("run show");
              strip.Show();
              Serial.println("executed");
              delay(1000);
              ledIndex = 0;
            }
           }
  }
 
  delay(1000);
  }
}

```
this is the Serial monitor leds on and it stays the same when the leds are turning off:

```
200{"anObject":[array shorted]}
run show
executed
run show
executed
200{"anObject":[array shorted]}
run show
executed
run show
executed
200{"anObject":[array shorted]}
run show
executed
run show
executed
```

I tried  it with the neo Pixel brightness bus and also with the normal neopixel bus

Makuna, 2019-04-07:
Please try to follow the Issue report template.  You did not provide information as to what board you are using, what versions of anything.

What do you mean by getting dark?  Be specific.  Are they fading or just jumping?  Do you consider off dark or does it still have something just not very bright.

What format is the data being sent as (what do the numbers for pixels mean).
In other words, what are you doing on this line and what are you expecting the value you are passing to HtmlColor description to be?  Why are you using HtmlColor?
```
int htmlColor = anObject[i];
            strip.SetPixelColor(ledIndex, HtmlColor(htmlColor)); 
```

Have you debugged to make sure the color was set into the strip (by calling GetPixelColor and checking if they match)?

You may want to rework the use of `ledIndex`.  There is no need for it to be global like you have it and there is many assumptions on its values.  It makes it hard to understand what you doing.

Have you checked your runtime memory and made sure you are not running out of memory causing other issues?

chekaaaaa, 2019-04-08:
Ohh i am Sorry i forgot it. I am running an esp32 dev board. And the newest version of the build tools and arduino ide. 

The pixels are completely off/black/dark/rgb(0,0,0) 

The numbers are originally in this format 0xff00ff but when send over http my backend it is converted to  the decimal equivalent in this example "16711935" (0xff00ff==16711935). So its a html web based color but as decimal number. I just can say, it works.

The first line of your comment saves the value to an variable. Which is assigned to an LED with the index of "ledIndex" and because its a Hex Color and not rgb i use the HTML Color object.

The color is set correctly because the first few minutes everything works as expected. 

How can I check if I run out of memory? But I dont think thats the problem because the sketch runs further and prints every two second the new response from my database. Only the LEDs are turning off. 

I hope you understand it now better ;) Thank you for your help

Makuna, 2019-04-08:
Again, be specific, not general.  What is the versions (newest is not specific)?  What EXACT libraries are you using for JSON?

If you remove the WiFi code completely and have a static color table that sets the colors in same way; does the problem reproduce?

Running out of memory can cause strange effects.  You need to do homework to figure out how to do check on your dynamic heap and stack space.  I would suggest asking about memory methods [on the Esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32).

You can't ASSUME because it works for bit, it is working in all cases.  You need to drop some debug output inplace and CONFIRM that the colors being set are correct when the problem appears; and that they are being set to the correct index (so output the pixel number you are thinking you are setting).

chekaaaaa, 2019-04-09:
The last part of your comment lead me to the problem. I printed out the heap size on every strip.Show and on every cycle of the program the heap got smaller. Until it hits a minimal threshold. The LEDs stop working, the heap is not getting lower anymore. It stays on the minium. Now i have to find a way to free up the heap after every cycle. Maybe you have an idea ;) But now I know it is not related to your library. I am sorry ;(. 

Thank you for you help ;)

Makuna, 2019-04-09:
Check the JSON library use pattern.  You may need to free something that it returns or call something on it to free.  
When ever I see this pattern of coding used I get worried, as it is not a recommended pattern in C++.
```
JsonObject& root = jsonBuffer.parseObject(payload);
```

closed 2019-04-09 by Makuna