Makuna/NeoPixelBus #268: LGPL is awkward on an embedded system...

owendelong, 2019-04-25:
**Is your feature request related to a problem? Please describe.**
The GNU LGPL that the library is currently licensed under has great intent, but for embedded platforms is somewhat impractical. Specifically, provisions 4a and 4b are difficult at best. Provision 4d0 is nearly impossible and 4d1 as an alternative is completely impossible.

**Describe the solution you'd like**
Please consider releasing under an alternative license such as the BSD license or granting a license exception for use in embedded products where lack of an OS and memory constraints as well as limited packaging space and the fact that the GPL itself is 20 times as much text as the entire product document render it impractical.

**Describe alternatives you've considered**
Short of a workable license, I'll probably have to resort to writing my own lesser library that meets my rather limited needs.

**Additional context**
Section 4 of the LGPL for reference (From COPYING):
  4. Combined Works.

  You may convey a Combined Work under terms of your choice that,
taken together, effectively do not restrict modification of the
portions of the Library contained in the Combined Work and reverse
engineering for debugging such modifications, if you also do each of
the following:

   a) Give prominent notice with each copy of the Combined Work that
   the Library is used in it and that the Library and its use are
   covered by this License.

   b) Accompany the Combined Work with a copy of the GNU GPL and this license
   document.

   c) For a Combined Work that displays copyright notices during
   execution, include the copyright notice for the Library among
   these notices, as well as a reference directing the user to the
   copies of the GNU GPL and this license document.

   d) Do one of the following:

       0) Convey the Minimal Corresponding Source under the terms of this
       License, and the Corresponding Application Code in a form
       suitable for, and under terms that permit, the user to
       recombine or relink the Application with a modified version of
       the Linked Version to produce a modified Combined Work, in the
       manner specified by section 6 of the GNU GPL for conveying
       Corresponding Source.

       1) Use a suitable shared library mechanism for linking with the
       Library.  A suitable mechanism is one that (a) uses at run time
       a copy of the Library already present on the user's computer
       system, and (b) will operate properly with a modified version
       of the Library that is interface-compatible with the Linked
       Version.

   e) Provide Installation Information, but only if you would otherwise
   be required to provide such information under section 6 of the
   GNU GPL, and only to the extent that such information is
   necessary to install and execute a modified version of the
   Combined Work produced by recombining or relinking the
   Application with a modified version of the Linked Version. (If
   you use option 4d0, the Installation Information must accompany
   the Minimal Corresponding Source and Corresponding Application
   Code. If you use option 4d1, you must provide the Installation
   Information in the manner specified by section 6 of the GNU GPL
   for conveying Corresponding Source.)

owendelong, 2019-04-25:
I understand, appreciate, and support the intent of the LGPL. Just in this case, the specific terms make literal compliance nearly impossible.

Makuna, 2019-04-25:
Part of my code is from the Adafruit NeoPixel code base, so I am restricted to use what they use.

owendelong, 2019-07-11:
So where I've ended up as a result of this is that I will end up re-implementing the library using my own code to at least remove the Adafruit code (they're unwilling to offer non-GPL terms).

My intent is to release the new library under the BSD license (free to use, free to modify, free to incorporate into projects/products as you see fit, but no forced-open-source of "contaminated" code).

I'd like your permission to reuse your code where I can separate it from the Adafruit code under those terms, but I understand if you're unwilling to grant it. Please advise.

rdragonrydr, 2019-08-15:
Would it be feasible to provide a link for the user to visit a website where they can get the code and license? That could count as "attached."

I don't think I've ever seen an application that _literally_ bundled the source code(s) with it...

Makuna, 2019-08-16:
@owendelong I do NOT grant permission to incorporate any part of this code into any project that is sold commercially.  Contact me directly if this is something you need to do.

owendelong, 2019-08-31:
Technically, the LGPL allows me to do that already, though it requires me to make the source to the library (and minimal linkable object code) available.

Happy to contact you directly and discuss further if you will provide contact information.

It's the minimum linkable object code that's the difficulty as it's pretty hard to release a linkable .o file for an arduino-IDE based project.

owendelong, 2019-09-04:
To clarify, what I'd like to do is take your code, remove the Adafruit derived code and replace it with my own code, and upload that into a new repository under BSD License Terms.

I will then use that (open source) library in creating my commercial (primarily hardware) product.

Obviously, to do that, I need your permission to release the code I copied from you under the BSD license. I'm not asking for permission to release a closed-source copy of your code in a commercial product.

Makuna, 2019-09-04:
The issue is that even my code has parts based on work done by others, so without know exactly what parts I can't agree.  Some of this was minimally annotated, this is my concern about a comercial product.

owendelong, 2019-09-05:
Understood. For now, it looks like I'll be trying to hand-compile a releasable .o file that people can download if they really want to. Later, I'll probably have to scratch-build a library with inspiration and "fair use".

owendelong, 2019-09-09:
Took some doing, but I figured out how to extract a .o from the Arduino IDE Compilation process.

Makuna, 2019-09-09:
would you be willing to share the process by documenting it in the Wiki?

owendelong, 2019-09-09:
I just posted it on Stack Exchange. I can also place it there.

owendelong, 2019-09-09:
Posted. Feel free to merge.