Makuna/NeoPixelBus #269: compilation Error: Neon Pixel Test

FerdinandFidel, 2019-04-25:
**Describe the bug**
When I upload the Neon Pixel Test on ESP8266 I get compilation errors.

**To Reproduce**
Steps to reproduce the behavior:
1. GitHub and download the lib
2. Open the NeonPixelTest.ino and compile it with Arduino 1.8.8,NeonPixelBus lib v.2.4.4

**Expected behavior**
No compiler errors 

**Development environment (please complete the following information):**
 - OS: [Mac OS 10.14]
 - Build Environment [Arduino IDE v.1.8.8]
 - Board target [Node MCU 1.0 (ESP-12 Modul)]
 - Library version [v2.4.4] 

**Minimal Sketch that reproduced the problem:**
See the sketch from NeonPixelTest.ino

**Debug Message**
Arduino: 1.8.8 (Mac OS X), Board: "NodeMCU 1.0 (ESP-12E Module), 160 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

Build-Optionen wurden verändert, alles wird neu kompiliert
In file included from /Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/Arduino.h:38:0,
                 from sketch/NeoPixelTest.ino.cpp:1:
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h: In member function 'void NeoEsp8266DmaMethodBase<T_SPEED>::Initialize()':
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:32:134: error: there are no arguments to 'rom_i2c_writeReg_Mask' that depend on a template parameter, so a declaration of 'rom_i2c_writeReg_Mask' must be available [-fpermissive]
 #define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
                                                                                                                                      ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:33:55: note: in expansion of macro 'i2c_writeReg_Mask'
 #define i2c_writeReg_Mask_def(block, reg_add, indata) i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
                                                       ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:756:43: note: in expansion of macro 'i2c_writeReg_Mask_def'
 #define I2S_CLK_ENABLE()                  i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1)
                                           ^
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h:262:9: note: in expansion of macro 'I2S_CLK_ENABLE'
         I2S_CLK_ENABLE();
         ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:32:134: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
 #define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
                                                                                                                                      ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:33:55: note: in expansion of macro 'i2c_writeReg_Mask'
 #define i2c_writeReg_Mask_def(block, reg_add, indata) i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
                                                       ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:756:43: note: in expansion of macro 'i2c_writeReg_Mask_def'
 #define I2S_CLK_ENABLE()                  i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1)
                                           ^
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h:262:9: note: in expansion of macro 'I2S_CLK_ENABLE'
         I2S_CLK_ENABLE();
         ^
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h: In instantiation of 'void NeoEsp8266DmaMethodBase<T_SPEED>::Initialize() [with T_SPEED = NeoEsp8266DmaSpeedWs2812x]':
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/NeoPixelBus.h:136:9:   required from 'void NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::Begin() [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x>]'
/Users/sebastian/Downloads/NeoPixelBus-master/examples/NeoPixelTest/NeoPixelTest.ino:76:17:   required from here
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:32:134: error: 'rom_i2c_writeReg_Mask' was not declared in this scope
 #define i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)  rom_i2c_writeReg_Mask(block, host_id, reg_add, Msb, Lsb, indata)
                                                                                                                                      ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:33:55: note: in expansion of macro 'i2c_writeReg_Mask'
 #define i2c_writeReg_Mask_def(block, reg_add, indata) i2c_writeReg_Mask(block, block##_hostid,  reg_add,  reg_add##_msb,  reg_add##_lsb,  indata)
                                                       ^
/Users/sebastian/Library/Arduino15/packages/esp8266/hardware/esp8266/2.5.0-beta3/cores/esp8266/esp8266_peri.h:756:43: note: in expansion of macro 'i2c_writeReg_Mask_def'
 #define I2S_CLK_ENABLE()                  i2c_writeReg_Mask_def(i2c_bbpll, i2c_bbpll_en_audio_clock_out, 1)
                                           ^
/Users/sebastian/Documents/Arduino/libraries/NeoPixelBus_by_Makuna/src/internal/NeoEsp8266DmaMethod.h:262:9: note: in expansion of macro 'I2S_CLK_ENABLE'
         I2S_CLK_ENABLE();
         ^
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

Makuna, 2019-04-25:
Please retry with the non-beta version of board support.  Esp8266 v2.5.0 was released and you are using v2.5.0beta3.

FerdinandFidel, 2019-04-25:
Yes this works! Thanks a lot & have a nice evening!

closed 2019-04-26 by FerdinandFidel