Makuna/NeoPixelBus #270: Esp32: Support I2s 24 parrallel mode

Makuna, 2019-06-06:
**Is your feature request related to a problem? Please describe.**
Esp32 i2s supports a 24bit parrellel mode; where each bit of 3 byte packet is sent to different pins.  This could be exploited to support 24 parrallel NeoPixel busses.  

**Describe the solution you'd like**
FastLED has support for this.  
[Code is here](https://github.com/samguyer/FastLED)
[Discussion is here](https://www.reddit.com/r/FastLED/comments/bjq0sm/new_24way_parallel_driver_for_esp32/)

**Additional context**
Currently using this method will require each bus to be same count and pixel data width (RGB or RGBW). 
FastLED requires this to be the same color element order but this is a minor limitation in their implementation.

r1dd1ck, 2019-06-16:
**This would be great** :thumbsup: 

But I'm somewhat concerned about the performance hit this will most probably take, as the update routine is **a lot heavier** than what is required by all the other methods :neutral_face:

And I'm not sure I understand the reasons behind your list of limitations :hushed:

**Same pixel count?** 
Can't you just fill the data for the "shorter strips" with **zeroes** to match the bus with most pixels?

**Same pixel data width?**
Filling the last byte of the 24bit (RGB) vs 32bit (RGBW) pixel difference with zeroes will result in a 10μs LOW period on the output between each pixel. This should be not enough to cause a RESET, so I think it should be possible to solve even this "issue".

At this point, the only plausible limitation, to me, seems to be the overall **update rate** (or frame rate) - which would need to be the same across all channels. But with a little effort, I'm sure that even that could be solved. Although not very sure whether it would it..

hpwit, 2019-10-22:
in theory it will be possible with the same code to drive RGBW and RGB over the 24 pins
I just did not wrote it yet. but I could look at it

Makuna, 2019-10-23:
(currently still capturing needs and requirements, don't let my original comments stop you from providing good feedback).

embedded-creations, 2021-12-13:
I'm starting to think about how to implement this.  The main challenge I see is allowing each bus to be used independently, though because all busses are shifted out in parallel (calling `bus0.show()` would shift out the pixels for all busses, not just `bus0`).  It wouldn't be a problem to call `bus0.show()` (which shifts out the pixels) followed by `bus1.show()` (which should do nothing) if the I2S Parallel method resets the dirty bit on all the busses using I2S Parallel.  I don't see a good way to call `ResetDirty()` for all busses from a method though.

Another challenge is initializing the I2S peripheral appropriately for the number of parallel strips and the length of the strips.  If treating each bus as independent as possible, calling `bus0.Begin()` should initialize the I2S peripheral, but the method won't have enough information on how many strips will be used, or the number of bytes in the longest strip.  We could create three separate methods for 8, 16, and 24 bit parallel.  We could limit the longest strip to the length of the first strip that calls Begin().  Or, we could require calling `SetMethodSettings` with a new SettingsObject passing that information to the method before Begin().  More challenging would be dynamically resizing the buffers and reconfiguring the I2S peripheral as new busses are added.

Thoughts @Makuna?

hpwit, 2021-12-13:
Hello in the fastLED library I have implemented the 24 way i2s parallel bus for esp32. 
i have reimplemented it in this https://github.com/hpwit/I2SClocklessLedDriver
But only for 16 // because if you go more than 16 you do not have enough pin left for extra stuff on the esp. 
hence I have done virtual pins driver which allows you to drive 120 // pin thanks to shift registers 
https://github.com/hpwit/I2SClocklessVirtualLedDriver

Let me know what you think 
Yves

embedded-creations, 2021-12-13:
@hpwit Thanks for the links Yves!  I knew you wrote the FastLED driver but didn't know you had separate repositories with lots of documentation too.  What is the license on your code?  I'm assuming MIT license as you contributed to the MIT License FastLED project, but it would be helpful if you added a license file to your repositories to be sure.

Makuna, 2021-12-13:
@embedded-creations This issue with using a single bus (and thus removing the confusing dirty and show issues) is that they would then use a single type of LED feature.  This sort of feature was tracked by https://github.com/Makuna/NeoPixelBus/issues/160.   Espresif has documented but not supplied/exposed a lag fix for all 8 channels of RMT.  It would require a single "Show" solution from my library; so I have been thinking about how to fix this internally.  And my current design is not a monolithic bus, but a new method series for ESP32 would be exposed that share a static (think global) dirty state and a single show will trigger all (first declared bus or any show triggers is still a design consideration).

@hpwit So in reality we have two implementations.  
* Use the inherent 24 parallel output of the ESP32 I2s. 
* Use an external shift register(s) to convert the serial stream to n channels of parallel output. 

ESP32 I2s Parallel output:  
* I believe this mode requires 24 bits always.  So, there is memory waist for channels unused and for any single bus less than the longest length of a bus.

Shift Register Output:
* How do you sync the serial latch and clock with the i2s stream?  Are you still using the i2s parallel output?

embedded-creations, 2021-12-13:
> I believe this mode requires 24 bits always

It doesn't, I use I2S parallel output in both 8bit and 16bit mode in [SmartMatrix Library](https://github.com/pixelmatix/SmartMatrix/blob/master/src/MatrixEsp32Hub75Calc_Impl.h#L720).  

> Q: Is the clock really just a bit in the stream

I've looked at Yves implementation before, clock is the I2S Clock output, not a data output.  Latch is a data output, that's why only the total output is 8x15=120 strips with 16-bit I2S.

Makuna, 2023-01-17:
Working Branch:  https://github.com/Makuna/NeoPixelBus/tree/Esp32I2sX
Current Status, tested for ESP32 for single and 8 bit.
Next will be testing for ESP32S2 modules.

hpwit, 2023-01-17:
Nice let me know how it goes.

Makuna, 2023-01-27:
https://github.com/Makuna/NeoPixelBus/tree/Esp32I2sX
Will be merging this into the main branch within the week.
Supports ESP32 and ESP32S2 (C3 and S3 can't use i2s for parallel mode, they have a new LED peripheral).
Supports x8 and x16 channel modes.  I didn't even try 24 bit mode but I am unsure more than 16 will ever be used and testing matrix was already taken a long time due to S2 being different let along i2s0 and i2s1 on ESP32 being different.

Makuna, 2023-01-27:
Merged in.  If someone every really needs x24 then create an new issue and I will look into it.  Is there even 24 pins available?

hpwit, 2023-01-28:
Hello 
You have only 22 outpins on the regular esp32.  I did the same limit to 16 pins output. I have made the i2svirtualclocklessleddriver that allows you,using shift registers, to driver 8 // strips per esp32 pins. And this over 15pins. Hence you can drive 120 strips in parrallel. I use this for my panel. I use 4 pins to drive 16 strips (clock, latch, 2 x data) so I have pins available to drive sd card, i2c Nintendo controller ….

Makuna, 2023-02-03:
https://github.com/Makuna/NeoPixelBus/releases/tag/2.7.2

closed 2023-02-03 by Makuna