Makuna/NeoPixelBus #275: Does not compile on ESP8266, does compile on ESP32

probonopd, 2019-06-18:
**Describe the bug**
Fails to compile on ESP8266, works on ESP32

**To Reproduce**
Steps to reproduce the behavior:
1. Compile https://github.com/probonopd/WirelessPrinting/blob/9387a641521c4083b859f89b3d3a5cb672c395bc/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino for ESP8266
2. See error

**Expected behavior**
Compiles

**Development environment (please complete the following information):**
 - OS: Linux
 - Build Environment: arduino-1.8.8-linux64.tar.xz
 - Board target: D1 mini
 - Library version: 2.5.0 and 2.4.4

**Minimal Sketch that reproduced the problem:**
https://github.com/probonopd/WirelessPrinting/blob/9387a641521c4083b859f89b3d3a5cb672c395bc/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino

**Additional context**
Full log:
https://travis-ci.org/probonopd/WirelessPrinting/builds/546989692?utm_source=github_status&utm_medium=notification

```

In file included from /home/travis/build/probonopd/WirelessPrinting/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino:21:0:
/home/travis/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h: In instantiation of 'void NeoPixelBus<T_COLOR_FEATURE, T_METHOD>::Show(bool) [with T_COLOR_FEATURE = NeoGrbFeature; T_METHOD = NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x>]':
/home/travis/build/probonopd/WirelessPrinting/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino:956:14:   required from here
/home/travis/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h:156:9: error: no matching function for call to 'NeoEsp8266DmaMethodBase<NeoEsp8266DmaSpeedWs2812x>::Update(bool&)'
         _method.Update(maintainBufferConsistency);
         ^
/home/travis/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h:156:9: note: candidate is:
In file included from /home/travis/Arduino/libraries/NeoPixelBus/src/NeoPixelBus.h:68:0,
                 from /home/travis/build/probonopd/WirelessPrinting/ESP8266WirelessPrintAsync/ESP8266WirelessPrintAsync.ino:21:
/home/travis/Arduino/libraries/NeoPixelBus/src/internal/NeoEsp8266DmaMethod.h:288:26: note: void NeoEsp8266DmaMethodBase<T_SPEED>::Update() [with T_SPEED = NeoEsp8266DmaSpeedWs2812x]
     void ICACHE_RAM_ATTR Update()
                          ^
/home/travis/Arduino/libraries/NeoPixelBus/src/internal/NeoEsp8266DmaMethod.h:288:26: note:   candidate expects 0 arguments, 1 provided
```

Makuna, 2019-06-18:
https://github.com/Makuna/NeoPixelBus/pull/276

probonopd, 2019-06-18:
Thank you very much, the sketch now can be compiled successfully for both processors.
https://travis-ci.org/probonopd/WirelessPrinting/builds/547442544

closed 2019-06-18 by probonopd