Makuna/NeoPixelBus #278: Default DMA mode unreliable, but BitBang and UART both fine

dwalker-uk, 2019-07-12:
Not sure if this is a bug, but keen to understand if you’ve seen it anywhere else...

Basically, doing some early testing before I finalise and get some (hobby) PCBs made, I had 3x strips of 8 WS2812B neopixels.  One was running DMA on the RX pin (GPIO3), one was on UART1 (GPIO 2) and the third was in BitBang mode on GPIO 14.  This is all on an ESP8266-12S.

I have a very simple sketch, which alternates the same pattern on all three LED strips (just the first 4 LEDs in each) every 3secs or so (actually every 100000 iterations through my loop()).  I only call Show() every 3 secs, when I want the pattern to change.  I left my sketch running, and noticed out of the corner of my eye that one of the strips frequently failed to update with the others - it just stuck on its last setting.  That was the DMA connected strip - it misses the update probably 1 in every 5 cycles, though its variable.  The other two are rock solid.  Nothing else is connected to the RX pin, and there’s no other notable activity (though it is connected to a WiFi network).

Any thoughts on possible causes?  I changed the Method for GPIO3 to BitBang, with no other changes, and it works perfectly like that.  However, I’m keep to understand if this is an oddity or something more fundamental, as I’d redesigned my board around keeping GOIO3 free for the LEDs.

Many thanks,

Dave

Makuna, 2019-07-12:
Please do not rely on BitBang; it really is outdated and will never been guaranteed to function reliably.

What is the declaration for the NeoPixelBus?  Specifically, which Method (exact text please) are you using?

Use the `NeoWs2812xMethod` not the  `NeoWs2812Method` method.

dwalker-uk, 2019-07-13:
Thanks for responding.

I was using NeoEsp8266Dma800KbpsMethod, when I had the issues.  I switched to NeoEsp8266BitBang800KbpsMethod which has been working fine, but as I noted that's without any WiFi activity and so understand your warning that I could run into trouble later.

I tried changing to NeoWs2812xMethod as suggested, but had the same problem - every few cycles it just completely misses an update.  For completeness I also tried NeoEsp8266DmaWs2812xMethod, but believe they're all essentially the same and so got the same result.

Within the Arduino IDE, I'm using Arduino 1.8.9, NeoPixelBus 2.5.0, "esp8266 by ESP8266 Community" 2.5.0.  I'm using the "Generic ESP8266 Module".

There's currently all sorts of other code in there, but I'll go away now and create a blank sketch to eliminate me having done anything else silly somewhere that could be affecting it in the sketch... will report back shortly!

dwalker-uk, 2019-07-13:
Ok I've identified the problem...

Basically, DMA mode on its own works fine for a single strip.  DMA mode with UART works perfect for two strips.  But if I want three separate strips, I cannot use DMA at the same time as BitBang.

In any combination, if I have both DMA and BitBang running at the same time, BitBang works fine but it seriously messes up whatever's connected to the DMA pin.  Minimal example code below.  Strangely, without my other code in there (presumably slowing it down??) the artefacts on the DMA pin are much worse - previously it was always the right colours and just missed some updates.  Now it's missing updates and sometimes doing random colours!

I don't expect to use multiple strips too often, so now I know the cause it's not too much trouble to just avoid.  The only reason for doing it was that if I did need more than 3 strips, I was planning to put the longest strip on DMA, the second and third on the UARTs, and any additionals on other pins with BitBang.  I assumed that could give maximum stability for the longer strips and that BitBang would be more tolerant on shorter strips.  But maybe not...?

Interested to understand whether others can replicate this, or if there's something odd about my setup.

Thanks a lot!

`#include <NeoPixelBus.h>

#define colorSaturation 50
RgbColor red(colorSaturation, 0, 0);
RgbColor green(0, colorSaturation, 0);
RgbColor blue(0, 0, colorSaturation);
RgbColor white(colorSaturation);
RgbColor black(0);

const uint16_t PixelCount = 8;
NeoPixelBus<NeoGrbFeature, NeoEsp8266DmaWs2812xMethod> strip1(PixelCount, 3);
NeoPixelBus<NeoGrbFeature, NeoEsp8266BitBang800KbpsMethod> strip2(PixelCount, 2); // Was NeoEsp8266Uart1800KbpsMethod...

int loopnum = 0;

void setup() {
  strip1.Begin();
  strip1.Show();
  strip2.Begin();
  strip2.Show();
}

void loop() {
  loopnum ++;
  if (loopnum == 100000) {
    strip1.SetPixelColor(0, red);
    strip1.SetPixelColor(1, green);
    strip1.SetPixelColor(2, blue);
    strip1.SetPixelColor(3, white);
    strip1.Show();
    strip2.SetPixelColor(0, red);
    strip2.SetPixelColor(1, green);
    strip2.SetPixelColor(2, blue);
    strip2.SetPixelColor(3, white);
    strip2.Show();
  }
  else if (loopnum == 200000) {
    strip1.SetPixelColor(0, black);
    strip1.SetPixelColor(1, red);
    strip1.SetPixelColor(2, black);
    strip1.SetPixelColor(3, red);
    strip1.Show();
    strip2.SetPixelColor(0, black);
    strip2.SetPixelColor(1, red);
    strip2.SetPixelColor(2, black);
    strip2.SetPixelColor(3, red);
    strip2.Show();
    loopnum = 0;
  }
}`

Makuna, 2019-07-14:
Something to try, wait on the dma bus instance until it finished, then call show on the bitbang bus instance.  DMA is async and relies on interrupts to manage the dma; the bitbang will suspend all interrupts in an attempt to stop anyone from effecting the timing sensitive data format.   If the DMA is still sending (again, the show will return before its finished sending the data) when the show on the bitbang is called, it can cause problems.

Makuna, 2019-08-04:
Closing this, as the bitbang should be avoided as documented in the Wiki due to interrupt suspending causing issues.
Three strips can be done on esp8266, you will need to use both uarts and the dma to accomplish; which means you will not have debug serial.

closed 2019-08-04 by Makuna