Makuna/NeoPixelBus #282: Shut down an initialized NeoPixelBus object on ESP8266?

rdragonrydr, 2019-08-15:
**Is your feature request related to a problem? Please describe.**
I've had occasional issues when programming an ESP8266 project that uses this library. What happens is that the DMA driver(s) causes LED-related code to appear on the stack dump, even if I'm not actually running any code at the moment that had to do with the Neopixels I'm controlling.
There are also times that it might be nice to choose between the onboard UART RX and the Neopixel control (since the CP2102 is hardwired to that pin), or to otherwise shut off an instance of the library.

**Describe the solution you'd like**
I would like a counterpart to the Begin() function, perhaps called Stop() or Halt(), that undoes any and all strand initialization, disabling DMA and ideally also letting me use the pins as normal.
I can't make heads or tails of the code to do this myself, since it's templated and thus confusing for me to find the relevant functions, and I don't know how the DMA drivers are actually functioning to know what flags to set to reset it to pre-Begin status.

Makuna, 2019-08-16:
You can dynamically allocate the object, and delete it.  When you delete it, it will wait for all relevant async work to finish, free memory, and then release pins for other uses.

```
#typedef NeoPixelBus<NeoFeatureRgb, Neo800MbpsMethod> NeoBus

NeoBus* strip = NULL; // pointer to strip instance, but non allocated yet
setup() { 
   strip = new NeoBus(3,2); // create one passing the number of pixels and the pin to use 
   strip->Begin(); // its a pointer, so must use the -> to access methods off the instance
}

void someOtherCode() {
  delete strip; // stop the bus; this will block if there is any pending async sending going on; but not for long unless you have an extremely high count of pixels (don't work about 1000 pixels)
}
```

Makuna, 2019-08-16:
Further, if all you want to do is make sure all async work is finished, you can just wait with something like this...
```
        while (!strip.CanShow())
        {
            yield();
        }
        // at this point, there are no pending async activity on the pin
```

closed 2019-09-28 by Makuna