Makuna/NeoPixelBus #289: Missing hash function

anders0nmat, 2019-09-15:
Hi, i try to use your RgbwColor in a std::unordered_Map as Key. The compiler complains about something with hashes so i guess there is a hash function missing.

`hashtable_policy.h: 1072:12: error: invalid use of incomplete type 'struct std::hash<RgbwColor>`

Makuna, 2019-09-15:
You did not seek help on the gitter channel first.
You did not follow the template for creating issues.
You did not do reasonable research for using std::unordered_map.
You assumed everyone HAS to directly support std containers to be able to be used by std containters.
Please do your homework next time.
https://stackoverflow.com/questions/17016175/c-unordered-map-using-a-custom-class-type-as-the-key

closed 2019-09-15 by Makuna

anders0nmat, 2019-09-15:
1. Ok, I didnt see that
2. Thats because i felt that my "question" doesnt fit in those two categories
3. I did afterwards. I know, shame on me.
4. Kind of. I just assumed Im missing something
I will
Yeah, thats where i found it too.
in case you're interested i wrote a Hashing class which just casts the RGBW to Int and uses already implemented Hash functions

Makuna, 2019-09-21:
This library is an Arduino library, as such, STL is not available on many platforms and thus isn't used within it.

As the link stated, just add something like one of the following to make it work.
```
// for RgbColor
namespace std {
  template <>
  struct hash<RgbColor>
  {
    std::size_t operator()(const RgbColor& k) const
    {
      return (k.R << 16 | k.G << 8 | k.B);
    }
  };
}
```
or
```
// for RgbwColor
namespace std {
  template <>
  struct hash<RgbwColor>
  {
    std::size_t operator()(const RgbwColor& k) const
    {
      return (k.W << 24 | k.R << 16 | k.G << 8 | k.B);
    }
  };
}
```