Makuna/NeoPixelBus #290: LPD8806 Support

MetricsGaming, 2019-09-19:
I am trying to integrate the LPD8806 LED strips with WLED. I currently have an existing Hyperion setup dependent on these strips.

I see that the APA102 with the data pin and clock pin is supported. The timings are a little different then LPD8806. Is there a way to add LPD8806 compatibility as well, instantly making these strips with WLED viable?

I have tried using the APA102 to see if it "just worked" and it unfortunately doesn't. Brightness doesn't work and the colors do not matchup. The preset scenes actually seem to kind of work but the colors are all over the place as well. I presume it's because of the mismatch in timings. The other alternative is ripping out the old strips for supported ones, the wife does not approve of that.

Makuna, 2019-09-19:
It can be done, but I need a protocol spec sheet.  It seems to be elusive/proprietary.
I also do not have any of these to test with.

This is the best I have found, which really isn't helpful; and even seems to have conflicting data.

https://cdn-shop.adafruit.com/datasheets/lpd8806+english.pdf

In the general section it states up to 20MHhz clk rate, but in the recommended it states 1-2 MHz!  

What platform (AVR, Esp8266, etc) are you using?

The hardware SPI defaults to 20mhz, you modify [this line ](https://github.com/Makuna/NeoPixelBus/blob/d83c0c7088e1fa9f4cfc2ab916a2951be5f8b371/src/internal/DotStarSpiMethod.h#L68) to 
```
SPI.beginTransaction(SPISettings(2000000L, MSBFIRST, SPI_MODE0));
```
to switch it down to 2mhz and see if that improves things.

Makuna, 2019-09-19:
Ok, its more complex than just changing the SPI frequency.
It will require some color translation and special "latching" support; as it isn't even close to a APA102.
But Adafruit had some code for it.
https://github.com/adafruit/LPD8806/blob/master/LPD8806.cpp

MetricsGaming, 2019-09-21:
> It can be done, but I need a protocol spec sheet. It seems to be elusive/proprietary.
> I also do not have any of these to test with.
>
> This is the best I have found, which really isn't helpful; and [...]

Currently I am running a raspberry pi with ubuntu with hyperion installed on it. I want to be able to use and eps8266 chipset. I did purchase the strips from adafruit so it makes sense they have some support code with it. I have my 36 led test strip I could send you? I wonder if looking through the Hyperion code would help? [https://github.com/hyperion-project/hyperion](url) I do see what you are talking about in the data sheet. I did find some code where this was set #define DEFAULT_SPI_SPEED        2500000 [https://github.com/gkaindl/ambi-tv/tree/master/src/components](url)

Makuna, 2019-09-21:
Its faster if I order some.  Consider making a donation using the paypal link in the wiki/readme.

MetricsGaming, 2019-09-21:
> Its faster if I order some. Consider making a donation using the paypal link in the wiki/readme.

If that's faster easier, not a problem, it's done.

Makuna, 2019-09-22:
Do you know the color order of the strip that you have?  It seems most hard code to GRB order.

Makuna, 2019-09-23:
https://github.com/Makuna/NeoPixelBus/pull/291
My Lpd886 Strip showed up.  Even Adafruits old code didn't work well beyond 32 pixels, they didn't have the protocol correct.
for bitbang, declare your strip like...
```
NeoPixelBus<Lpd8806BrgFeature, Lpd8806Method> strip(PixelCount, DotClockPin, DotDataPin);
```
for SPI, declare your strip like...
```
// hardware SPI support often has hard coded pins, so no need to pass them in
NeoPixelBus<Lpd8806BrgFeature, Lpd8806SpiMethod> strip(PixelCount); 
```
Let me know if your color order is not supported, I only added two for now, Grb and Brg.  Mine were Brg.  You can use the DotStartTest sample and modify the constructor to confirm.  
NOTE:  Adjust the strip count, if you set it to less than you have with DotStars or these, the pixel after the last one you defined will light up white'ish.

You can copy the repo directly into your installed version to get the latest.  I will put out a new release in the next few days.

MetricsGaming, 2019-09-26:
> Do you know the color order of the strip that you have? It seems most hard code to GRB order.

Mine are BRG

MetricsGaming, 2019-09-26:
> #291
> My Lpd886 Strip showed up. Even Adafruits old code didn't work well beyond 32 pixels, they didn't have the protocol correct.
> for bitbang, declare your strip like...
>
> ``` [...]

First of all thank you for your quick work, you have some fast shipping. I was surprised you got the LED strips so fast hence why there was a lapse in checking in sooner, sorry.

I have not been able to get bitbang or SPI to change the colors on the strip in the DotStarTest. I changed the 2 lines your have above in the DotStarTest.ino, am i missing something more? In the serial monitor I can see the expected output, just not reflected on the strips. I changed the pin out to match the wiring on my nodemcu and pixel count to 36
DotClockPin = 0;
DotDataPin = 2;

Makuna, 2019-09-26:
If you use SPI (hardware) on most Arduino's then those are not the right pins, the pins you pass are ignored and the hardware defined pins are used.  The pins are dependant on the hardware.  
 Define = Description = Board Label
SCK = SPI Clock = Esp8266 Pin Labeled Clk, SCK, SPI_Clk, GPIO6
MOSI = SPI Data Out = Esp8266 Pin Labeled SPI_Mosi, GPIO8

Makuna, 2019-09-27:
This should now be in the released version

closed 2019-09-27 by Makuna